mkeeter/antimony #2: Installation problem on Ubuntu 14.04 LTS

Fablabsitges, 2015-01-22:
I followed all the steps but got stuck here. Do you have any idea of what could be the cause?

Thanks

irix@mba:~/Documents/code/antimony/build$ ~/Qt/5.4/gcc_64/bin/qmake ../qt/antimony.pro
fatal: no tag exactly matches '8196286bd2fc37accf53111bdf06338bf24b6e4d'

Fablabsitges, 2015-01-23:
I removed --exactmatch inside qt/antimony.pro and now it works

closed 2015-01-23 by Fablabsitges

mkeeter, 2015-01-23:
That error message isn't actually a cause for alarm -- it's used if there's an exact tag match to populate the release number in the About box -- but I'll look into hiding it (since you're the second person to be worried by it).

JMMolenaar, 2015-02-04:
Matt,

I'm getting  "collect2: error: ld returned 1 exit status"
when installing on Ubuntu 14.4
Any idea why?