mkeeter/antimony #3: CSG submenu does not appear if QT not installed in default location

Fablabsitges, 2015-02-04:
Hi Matt, we have 3 ubuntu machines here. 2 of them installed QT in the default folder (/Home), and another inside  /Home/Developer folder. This last machine cannot see the CSG submenu items. The first time he rollovers CSG, the submenu opens with the same ammount of items but empty. Then the second time rollover nothing happens. The rest of menus and submenus are OK.

To us it looks like if there might be a path somewhere in the code issue related to qt and the CSG menu but we cannot find it. Do you think it is possible?

mkeeter, 2015-02-05:
Are you running from the build directory or did you use `sudo make install` to drop it in `/usr/local/bin`?  It should be automatically populating nodes from the `sb/nodes` directory, either in the build directory or in `/usr/local/bin/sb/nodes`.

Try just running `make` again -- Neil had noticed on one machine that it only did the copying operation on the second run.

Fablabsitges, 2015-02-05:
Sorry for that. This machine was running antimony from outside the build directory. Running from inside populates the CSG menu.

Thank you Matt

closed 2015-02-05 by Fablabsitges