mkeeter/antimony #4: collect2: error - ld

JMMolenaar, 2015-02-05:
Matt,

I'm getting "collect2: error: ld returned 1 exit status"
when installing on Ubuntu 14.4
Any idea why? Could it be the placement of the library?

mkeeter, 2015-02-05:
Can you show me the complete output when you call `make`?

jacobron, 2015-02-12:
Hey, if it's saying 

/usr/bin/ld: cannot find -lgd
collect2: ld returned 1 exit status

get sure you have the following packages: mesa-common-dev libgd-dev
and run following command in your terminal creating a symlink to the libGL.so file (worked on x64 linux):

sudo ln -s /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1.2.0 /usr/lib/libGL.so

mkeeter, 2015-02-12:
Okay, this sounds like it's having trouble finding the OpenGL libraries.

What happens with you `apt-get install` the recommended packages and run the command that it recommended?  Does it still fail in the same way?

mkeeter, 2015-02-12:
@jacobron -- since you sent me a pull request with improved documentation, I assume that you've gotten this working.  Re-open if you're still having issues.

closed 2015-02-12 by mkeeter

JMMolenaar, 2015-02-12:
wow got completely sidetracked from this one.  I'll try your advice
@jacobron, thanks!

mottosso, 2015-06-06:
I'm also having this problem, having been unable to compile Antimony at all.

Here's the output of make, on Ubuntu 14.04 x64, a second time around (not being terribly familiar with make, assuming it's producing less output the second time and that it is relevant to the problem)

- https://gist.github.com/mottosso/57509ed0dd31f6b6c228

From what I can tell, the same problem occured even after installing `mesa-common-dev libgd-dev` and of course the suggested packages from the [build instructions](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md#linux).

Any ideas?

mkeeter, 2015-06-06:
@mottosso That's an interesting traceback; it almost looks like you were building only a part of the project.  Can you delete your build directory, recreate it, and make sure that you've called `qmake` on `antimony.pro` and not any of the other subprojects in the `qt` folder?

mottosso, 2015-06-06:
Sure. I think it's because I ran qmake once before, this traceback is from the second time I called it. So it's probably like you say, part of a build.

Trying to get a hold of the full traceback now.

mottosso, 2015-06-06:
Sorry, I meant that I ran `make -j4` once before, and that this is the second run.

mottosso, 2015-06-07:
Good news is, I got it working. Bad news is I don't know why it wasn't working at first.

I ran it on a clean install of Ubuntu 14.04 and didn't encounter any issues. I had already reset the VM by the time I posted the first traceback and was unable to test it again on that machine.