mkeeter/antimony #5: Compile problem with Ubuntu 12.04

GithubGeo, 2015-02-11:
This is my build log.
Thank you for your help,
Christian

/opt/Qt/5.4/gcc/bin/uic ../src/ui/forms/main_window.ui -o ui_main_window.h
/opt/Qt/5.4/gcc/bin/uic ../src/ui/forms/resolution_dialog.ui -o ui_resolution_dialog.h
/opt/Qt/5.4/gcc/bin/uic ../src/ui/forms/exporting_dialog.ui -o ui_exporting_dialog.h
gcc -c -pipe -std=gnu99 -O2 -Wall -W -D_REENTRANT -fPIE -DANTIMONY -D_STATIC_=  -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/opt/Qt/5.4/gcc/mkspecs/linux-g++ -I../qt -I../src/fab -I../src -I/opt/Qt/5.4/gcc/include -I/opt/Qt/5.4/gcc/include/QtOpenGL -I/opt/Qt/5.4/gcc/include/QtWidgets -I/opt/Qt/5.4/gcc/include/QtGui -I/opt/Qt/5.4/gcc/include/QtCore -I. -I. -I. -o home/geo/Programmi/antimony/src/fab/tree/eval.o ../src/fab/tree/eval.c
gcc -c -pipe -std=gnu99 -O2 -Wall -W -D_REENTRANT -fPIE -DANTIMONY -D_STATIC_=  -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/opt/Qt/5.4/gcc/mkspecs/linux-g++ -I../qt -I../src/fab -I../src -I/opt/Qt/5.4/gcc/include -I/opt/Qt/5.4/gcc/include/QtOpenGL -I/opt/Qt/5.4/gcc/include/QtWidgets -I/opt/Qt/5.4/gcc/include/QtGui -I/opt/Qt/5.4/gcc/include/QtCore -I. -I. -I. -o home/geo/Programmi/antimony/src/fab/tree/render.o ../src/fab/tree/render.c
gcc -c -pipe -std=gnu99 -O2 -Wall -W -D_REENTRANT -fPIE -DANTIMONY -D_STATIC_=  -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/opt/Qt/5.4/gcc/mkspecs/linux-g++ -I../qt -I../src/fab -I../src -I/opt/Qt/5.4/gcc/include -I/opt/Qt/5.4/gcc/include/QtOpenGL -I/opt/Qt/5.4/gcc/include/QtWidgets -I/opt/Qt/5.4/gcc/include/QtGui -I/opt/Qt/5.4/gcc/include/QtCore -I. -I. -I. -o home/geo/Programmi/antimony/src/fab/tree/tree.o ../src/fab/tree/tree.c
gcc -c -pipe -std=gnu99 -O2 -Wall -W -D_REENTRANT -fPIE -DANTIMONY -D_STATIC_=  -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/opt/Qt/5.4/gcc/mkspecs/linux-g++ -I../qt -I../src/fab -I../src -I/opt/Qt/5.4/gcc/include -I/opt/Qt/5.4/gcc/include/QtOpenGL -I/opt/Qt/5.4/gcc/include/QtWidgets -I/opt/Qt/5.4/gcc/include/QtGui -I/opt/Qt/5.4/gcc/include/QtCore -I. -I. -I. -o home/geo/Programmi/antimony/src/fab/tree/parser.o ../src/fab/tree/parser.c
gcc -c -pipe -std=gnu99 -O2 -Wall -W -D_REENTRANT -fPIE -DANTIMONY -D_STATIC_=  -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/opt/Qt/5.4/gcc/mkspecs/linux-g++ -I../qt -I../src/fab -I../src -I/opt/Qt/5.4/gcc/include -I/opt/Qt/5.4/gcc/include/QtOpenGL -I/opt/Qt/5.4/gcc/include/QtWidgets -I/opt/Qt/5.4/gcc/include/QtGui -I/opt/Qt/5.4/gcc/include/QtCore -I. -I. -I. -o home/geo/Programmi/antimony/src/fab/tree/triangulate.o ../src/fab/tree/triangulate.c
gcc -c -pipe -std=gnu99 -O2 -Wall -W -D_REENTRANT -fPIE -DANTIMONY -D_STATIC_=  -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/opt/Qt/5.4/gcc/mkspecs/linux-g++ -I../qt -I../src/fab -I../src -I/opt/Qt/5.4/gcc/include -I/opt/Qt/5.4/gcc/include/QtOpenGL -I/opt/Qt/5.4/gcc/include/QtWidgets -I/opt/Qt/5.4/gcc/include/QtGui -I/opt/Qt/5.4/gcc/include/QtCore -I. -I. -I. -o home/geo/Programmi/antimony/src/fab/tree/math/math_f.o ../src/fab/tree/math/math_f.c
gcc -c -pipe -std=gnu99 -O2 -Wall -W -D_REENTRANT -fPIE -DANTIMONY -D_STATIC_=  -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/opt/Qt/5.4/gcc/mkspecs/linux-g++ -I../qt -I../src/fab -I../src -I/opt/Qt/5.4/gcc/include -I/opt/Qt/5.4/gcc/include/QtOpenGL -I/opt/Qt/5.4/gcc/include/QtWidgets -I/opt/Qt/5.4/gcc/include/QtGui -I/opt/Qt/5.4/gcc/include/QtCore -I. -I. -I. -o home/geo/Programmi/antimony/src/fab/tree/math/math_i.o ../src/fab/tree/math/math_i.c
gcc -c -pipe -std=gnu99 -O2 -Wall -W -D_REENTRANT -fPIE -DANTIMONY -D_STATIC_=  -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/opt/Qt/5.4/gcc/mkspecs/linux-g++ -I../qt -I../src/fab -I../src -I/opt/Qt/5.4/gcc/include -I/opt/Qt/5.4/gcc/include/QtOpenGL -I/opt/Qt/5.4/gcc/include/QtWidgets -I/opt/Qt/5.4/gcc/include/QtGui -I/opt/Qt/5.4/gcc/include/QtCore -I. -I. -I. -o home/geo/Programmi/antimony/src/fab/tree/math/math_r.o ../src/fab/tree/math/math_r.c
../src/fab/tree/triangulate.c: In function ‘triangulate_voxel’:
../src/fab/tree/triangulate.c:111:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/triangulate.c:112:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/triangulate.c:113:35: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/tree.c: In function ‘clone_tree’:
../src/fab/tree/tree.c:82:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/tree.c:89:29: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/tree.c: In function ‘disable_nodes_binary’:
../src/fab/tree/tree.c:135:29: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/tree.c:136:25: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/tree.c:143:25: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/tree.c: In function ‘disable_nodes’:
../src/fab/tree/tree.c:173:29: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/tree.c:174:25: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/tree.c:188:25: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/tree.c: In function ‘active_axes’:
../src/fab/tree/tree.c:251:25: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c: In function ‘render8’:
../src/fab/tree/render.c:55:45: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:56:49: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:73:41: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:74:45: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c: In function ‘region8’:
../src/fab/tree/render.c:116:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:117:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:139:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:142:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c: In function ‘get_normals8’:
../src/fab/tree/render.c:175:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:178:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:180:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:183:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:185:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:188:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:191:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c: In function ‘shade_pixels8’:
../src/fab/tree/render.c:212:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c: In function ‘render16’:
../src/fab/tree/render.c:296:45: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:297:49: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:314:41: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c:315:45: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/fab/tree/render.c: In function ‘region16’:
[... cut ...]
../src/graph/datum/input.h:82:32: error: ‘override’ does not name a type
../src/graph/datum/input.h:83:25: error: expected ‘;’ at end of member declaration
../src/graph/datum/input.h:83:31: error: ‘override’ does not name a type
../src/graph/datum/input.h:84:29: error: expected ‘;’ at end of member declaration
../src/graph/datum/input.h:84:35: error: ‘override’ does not name a type
make: *** [home/geo/Programmi/antimony/src/graph/datum/datum.o] Errore 1
make: *** [home/geo/Programmi/antimony/src/graph/datum/types/eval_datum.o] Errore 1

mkeeter, 2015-02-11:
It looks like the compiler is having trouble on a bunch of `C++11` features, so I suspect it's outdated.
What version of `g++` do you have installed?  Can you upgrade it and try again?

GithubGeo, 2015-02-11:
Hi Matt,
i upgraded my g++ version at:
g++ (Ubuntu 4.9.2-0ubuntu1~12.04) 4.9.2

Building with this new version of g++  i found a new error:
/usr/bin/ld: cannot find -lboost_python-py34
/usr/bin/ld: cannot find -lpython3.4m
collect2: error: ld returned 1 exit status
make: *** [antimony] Errore 1
Thank you in advance of your help.

GithubGeo, 2015-02-11:
This is the output of the command:
ls /usr/lib/libboost_python*
/usr/lib/libboost_python.a           /usr/lib/libboost_python-py27.so
/usr/lib/libboost_python-mt-py27.a   /usr/lib/libboost_python-py27.so.1.46.1
/usr/lib/libboost_python-mt-py27.so  /usr/lib/libboost_python-py32.a
/usr/lib/libboost_python-mt-py32.a   /usr/lib/libboost_python-py32.so
/usr/lib/libboost_python-mt-py32.so  /usr/lib/libboost_python-py32.so.1.46.1
/usr/lib/libboost_python-py27.a      /usr/lib/libboost_python.so

GithubGeo, 2015-02-11:
I found this kind of solution for a similar problem:
sudo ln -s /usr/lib/libboost_python-mt-py26 /usr/lib/libboost_python-mt
We can adapt for the my problem?

jacobron, 2015-02-12:
looks like your libs are outdated - you should upgrade your distribution to 14.04

mkeeter, 2015-02-12:
Yes, it's looking for boost linked against Python 3.4 and you've only got Python 3.2.

prokoudine, 2015-05-14:
Got a similar issue:

/usr/bin/ld: cannot find -lboost_python-py34

Except I'm on Fedora 21, and there's only Python 3.4 here.

$ ls /usr/lib/libboost_python
libboost_python3.so         libboost_python3.so.1.55.0  libboost_python.so          libboost_python.so.1.55.0

Any ideas?

mkeeter, 2015-05-14:
I think this is due to a mismatch in library names.  Can you locally change [this line](https://github.com/mkeeter/antimony/blob/develop/qt/shared.pri#L44) to read `LIBS += -lboost_python` and see if that fixes it?

If so, we can figure out how to put a Fedora-specific case into the `qmake` file.

prokoudine, 2015-05-14:
I changed that line to

LIBS += -lboost_python -lpython3.4m

Here's what happens next:

home/alex/soft/cad/antimony/antimony/src/fab/fab.o: In function `PyInit__fabtypes':
/home/alex/soft/cad/antimony/antimony/build/../src/fab/fab.cpp:21: undefined reference to `boost::python::detail::init_module(PyModuleDef&, void (*)())'
home/alex/soft/cad/antimony/antimony/src/graph/node/proxy.o: In function `PyInit__proxy':
/home/alex/soft/cad/antimony/antimony/build/../src/graph/node/proxy.cpp:89: undefined reference to `boost::python::detail::init_module(PyModuleDef&, void (*)())'
home/alex/soft/cad/antimony/antimony/src/graph/hooks/hooks.o: In function `PyInit__hooks':
/home/alex/soft/cad/antimony/antimony/build/../src/graph/hooks/hooks.cpp:29: undefined reference to `boost::python::detail::init_module(PyModuleDef&, void (*)())'
collect2: error: ld returned 1 exit status
Makefile:624: recipe for target 'antimony' failed
make: *** [antimony] Error 1

mkeeter, 2015-05-14:
Ah, okay -- it looks like `libboost_python` is a `2.x` version.  Try `LIBS += -lboost_python3` instead.

prokoudine, 2015-05-14:
That worked :) Thank you!

mkeeter, 2015-05-15:
Okay, I've updated the BUILDING guide with this caveat.
I'm closing this PR because it has strayed incredibly far from the original issue.

closed 2015-05-15 by mkeeter