mkeeter/antimony #6: Missing nodes on the 'Add' menu on Mac

James-Fletcher, 2015-02-11:
Pulled most recent version of repository on 11/2/2015.

Built on both MacOS 10.10.2 and Ubuntu 14.04.  On the Ubuntu machines I get all of the latest node types such as 'Torus' etc under the 'Add->3D' menu.  On 2 different Mac's I get a working Antimony build but there are less node types available, e.g. 'Torus' is missing.

The torus_x _y _z.node files all exist in the build/sb folder on the Mac builds.

mkeeter, 2015-02-12:
For some reason, the nodes aren't correctly copied over the first time you call `make`.  
Run it once more and it would work (but I'll look into why this is happening).

mkeeter, 2015-02-12:
Fixed in `d9c5dee86c9dfab5a23f55f6d6523e350de1ac28`.

closed 2015-02-12 by mkeeter

James-Fletcher, 2015-02-12:
Thanks Matt, the second call to make has fixed it for me.

gustavolsson, 2015-11-19:
Just wanted to let you know that I pulled the latest version today and this problem is still present. I'm running Ubuntu 15.10. After running make again it works though :)