mkeeter/antimony #8: Crash when exporting to png

mereufrance77, 2015-02-12:
This is the crash log:

Process:         antimony [6316]
Path:            /Users/USER/Desktop/antimony.app/Contents/MacOS/antimony
Identifier:      com.impraxical.antimony
Version:         0.7.4
Code Type:       X86-64 (Native)
Parent Process:  launchd [146]
Responsible:     antimony [6316]
User ID:         501

Date/Time:       2015-02-12 15:48:34.703 +0100
OS Version:      Mac OS X 10.9.5 (13F34)
Report Version:  11
Anonymous UUID:  DDDE9FC4-CDE3-518E-06DF-D6B701E9CDEF

Sleep/Wake UUID: 7C389DEF-2E88-4204-ADF5-7C745C0188D1

Crashed Thread:  6  QThread

Exception Type:  EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes: 0x0000000000000001, 0x0000000000000000

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff8f8dfa1a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff8f8ded18 mach_msg + 64
2   com.apple.CoreGraphics        	0x00007fff8538cf1b _CGSSynchronizeWindowBackingStore + 97
3   com.apple.CoreGraphics        	0x00007fff8530e4ec _CGSLockWindow + 3765
4   com.apple.CoreGraphics        	0x00007fff8530d3ff CGSDeviceLock + 240
5   libRIP.A.dylib                	0x000000010e988487 ripd_Lock + 43
6   libRIP.A.dylib                	0x000000010e98e9b4 RIPLayerBltImage + 296
7   libRIP.A.dylib                	0x000000010e98e6aa ripc_RenderImage + 262
8   libRIP.A.dylib                	0x000000010e98cbff ripc_DrawImage + 1147
9   com.apple.CoreGraphics        	0x00007fff85321024 CGContextDrawImage + 457
10  libqcocoa.dylib               	0x000000010e854169 0x10e83d000 + 94569
11  com.apple.AppKit              	0x00007fff8945b0b1 -[NSView _drawRect:clip:] + 3846
12  com.apple.AppKit              	0x00007fff8945803c -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 3199
13  com.apple.AppKit              	0x00007fff89458bc4 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 6151
14  com.apple.AppKit              	0x00007fff89456eb1 -[NSThemeFrame _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 314
15  com.apple.AppKit              	0x00007fff89453e9f -[NSView _displayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:] + 2828
16  com.apple.AppKit              	0x00007fff894332da -[NSView displayIfNeeded] + 1680
17  com.apple.AppKit              	0x00007fff8949874e _handleWindowNeedsDisplayOrLayoutOrUpdateConstraints + 884
18  com.apple.AppKit              	0x00007fff89a6e061 __83-[NSWindow _postWindowNeedsDisplayOrLayoutOrUpdateConstraintsUnlessPostingDisabled]_block_invoke1331 + 46
19  com.apple.CoreFoundation      	0x00007fff8e2e4d67 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
20  com.apple.CoreFoundation      	0x00007fff8e2e4cd7 __CFRunLoopDoObservers + 391
21  com.apple.CoreFoundation      	0x00007fff8e2d63b8 __CFRunLoopRun + 776
22  com.apple.CoreFoundation      	0x00007fff8e2d5e75 CFRunLoopRunSpecific + 309
23  com.apple.HIToolbox           	0x00007fff8f233a0d RunCurrentEventLoopInMode + 226
24  com.apple.HIToolbox           	0x00007fff8f2337b7 ReceiveNextEventCommon + 479
25  com.apple.HIToolbox           	0x00007fff8f2335bc _BlockUntilNextEventMatchingListInModeWithFilter + 65
26  com.apple.AppKit              	0x00007fff892fc24e _DPSNextEvent + 1434
27  com.apple.AppKit              	0x00007fff892fb89b -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 122
28  libqcocoa.dylib               	0x000000010e85d8cd 0x10e83d000 + 133325
29  QtCore                        	0x000000010c9e05ad QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
30  QtWidgets                     	0x000000010be31362 QDialog::exec() + 514
31  com.impraxical.antimony       	0x000000010b824a8b App::onExportHeightmap() + 2795
32  QtCore                        	0x000000010ca155f2 QMetaObject::activate(QObject*, int, int, void**) + 1986
33  QtWidgets                     	0x000000010bc43eff QAction::activate(QAction::ActionEvent) + 271
34  QtWidgets                     	0x000000010bc44364 0x10bc1b000 + 168804
35  QtCore                        	0x000000010ca159db QMetaObject::activate(QObject*, int, int, void**) + 2987
36  libqcocoa.dylib               	0x000000010e861430 0x10e83d000 + 148528
37  com.apple.AppKit              	0x00007fff8951a260 -[NSApplication sendAction:to:from:] + 327
38  com.apple.AppKit              	0x00007fff895351c8 -[NSMenuItem _corePerformAction] + 394
39  com.apple.AppKit              	0x00007fff89534f04 -[NSCarbonMenuImpl performActionWithHighlightingForItemAtIndex:] + 117
40  com.apple.AppKit              	0x00007fff8958440d -[NSMenu _internalPerformActionForItemAtIndex:] + 35
41  com.apple.AppKit              	0x00007fff89584289 -[NSCarbonMenuImpl _carbonCommandProcessEvent:handlerCallRef:] + 104
42  com.apple.AppKit              	0x00007fff8952aff6 NSSLMMenuEventHandler + 716
43  com.apple.HIToolbox           	0x00007fff8f20d1d4 DispatchEventToHandlers(EventTargetRec*, OpaqueEventRef*, HandlerCallRec*) + 892
44  com.apple.HIToolbox           	0x00007fff8f20c787 SendEventToEventTargetInternal(OpaqueEventRef*, OpaqueEventTargetRef*, HandlerCallRec*) + 385
45  com.apple.HIToolbox           	0x00007fff8f220880 SendEventToEventTarget + 40
[... 411 lines cut ...]
__IMAGE                             528K
__LINKEDIT                         70.6M
__TEXT                            193.2M
__UNICODE                           544K
mapped file                        58.0M
shared memory                        68K
===========                      =======
TOTAL                               1.9G
TOTAL, minus reserved VM space      1.9G
 

Model: MacBookAir3,1, BootROM MBA31.0061.B07, 2 processors, Intel Core 2 Duo, 1.6 GHz, 4 GB, SMC 1.67f10
Graphics: NVIDIA GeForce 320M, NVIDIA GeForce 320M, PCI, 256 MB
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1067 MHz, 0x80CE, 0x4D34373142353237334348302D5946382020
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1067 MHz, 0x80CE, 0x4D34373142353237334348302D5946382020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xD1), Broadcom BCM43xx 1.0 (5.106.98.100.22)
Bluetooth: Version 4.2.7f4 15112, 3 services, 23 devices, 1 incoming serial ports
Network Service: AirPort, AirPort, en0
Serial ATA Device: APPLE SSD SM128C, 121,33 GB
USB Device: FaceTime Camera (Built-in)
USB Device: Composite Device
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Apple Internal Keyboard / Trackpad
Thunderbolt Bus:

mkeeter, 2015-02-13:
Does this happen anytime you try to export any image, or only sometimes?

fibasile, 2015-02-15:
As far as she told me any image with v. 0.7.4

mkeeter, 2015-02-15:
I'm having trouble reproducing this.  Can you send me an example file that crashes on `.png` export?

mereufrance77, 2015-02-16:
That's antimony files

 version OS : OSX 10.9.5

It crashed when exporting to png, any files..

mkeeter, 2015-02-17:
Can you give it a try with the latest release (`0.7.5`, from earlier today)?

mereufrance77, 2015-02-18:
I try with the latest release but it the same to export a png file. 
I attached a report:
Process:         antimony [15205]
Path:            /Applications/antimony.app/Contents/MacOS/antimony
Identifier:      com.impraxical.antimony
Version:         0.7.5
Code Type:       X86-64 (Native)
Parent Process:  launchd [146]
Responsible:     antimony [15205]
User ID:         501

Date/Time:       2015-02-18 09:23:39.112 +0100
OS Version:      Mac OS X 10.9.5 (13F34)
Report Version:  11
Anonymous UUID:  DDDE9FC4-CDE3-518E-06DF-D6B701E9CDEF

Sleep/Wake UUID: 17903CC2-8870-4FEE-B5C4-F114AE194354

Crashed Thread:  7  QThread

Exception Type:  EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes: 0x0000000000000001, 0x0000000000000000

Thread 0:: Dispatch queue: com.apple.main-thread
0   com.apple.coreui              	0x00007fff8d22afcb CUIElement::LoadFromArtFile(unsigned char const*, bool) + 801
1   com.apple.coreui              	0x00007fff8d22bcac CUIElement::Load(CUIDescriptor const*, long, long, long, bool, bool) + 124
2   com.apple.coreui              	0x00007fff8d1f9b1a CUIArtFileRenderer::DrawProgressIndicatorIndeterminate(CUIDescriptor const*) + 168
3   com.apple.coreui              	0x00007fff8d20b6af CUIArtFileRenderer::Draw(CUIDescriptor const*, CGAffineTransform, CUIReturnInfo&) + 1391
4   com.apple.coreui              	0x00007fff8d1e8988 CUIRenderer::Draw(CGRect, CGContext*, __CFDictionary const*, __CFDictionary const**) + 2904
5   com.apple.coreui              	0x00007fff8d21740a CUIDraw + 178
6   com.apple.HIToolbox           	0x00007fff8f2b34d7 _HIThemeDrawProgressBar(HIThemeTrackDrawInfo const*, ThemeEraseXUPP*, void*, CGContext*) + 246
7   com.apple.HIToolbox           	0x00007fff8f2b440f _HIThemeDrawTrackInternal(HIThemeTrackDrawInfo const*, CGRect const*, ThemeEraseXUPP*, void*, CGContext*, unsigned int) + 132
8   QtWidgets                     	0x00000001093fcbb6 0x1093e6000 + 93110
9   QtWidgets                     	0x00000001094ba897 0x1093e6000 + 870551
10  QtWidgets                     	0x000000010948c417 QCommonStyle::drawControl(QStyle::ControlElement, QStyleOption const*, QPainter*, QWidget const*) const + 2791
11  QtWidgets                     	0x00000001094de889 0x1093e6000 + 1017993
12  QtWidgets                     	0x00000001094b9cbe 0x1093e6000 + 867518
13  QtWidgets                     	0x000000010958f7bb QProgressBar::paintEvent(QPaintEvent*) + 123
14  QtWidgets                     	0x00000001094574bb QWidget::event(QEvent*) + 2123
15  QtWidgets                     	0x000000010958ff00 QProgressBar::event(QEvent*) + 176
16  QtWidgets                     	0x0000000109417dbb QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
17  QtWidgets                     	0x000000010941b110 QApplication::notify(QObject*, QEvent*) + 8192
18  QtCore                        	0x000000010a1b1fb3 QCoreApplication::notifyInternal(QObject*, QEvent*) + 115
19  QtWidgets                     	0x0000000109450c20 QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 3152
20  QtWidgets                     	0x000000010945154a QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 1226
21  QtWidgets                     	0x0000000109450fb0 QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 4064
22  QtWidgets                     	0x000000010942464d 0x1093e6000 + 255565
23  QtWidgets                     	0x0000000109423666 0x1093e6000 + 251494
24  QtWidgets                     	0x0000000109471508 0x1093e6000 + 570632
25  QtWidgets                     	0x0000000109417dbb QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
26  QtWidgets                     	0x000000010941b110 QApplication::notify(QObject*, QEvent*) + 8192
27  QtCore                        	0x000000010a1b1fb3 QCoreApplication::notifyInternal(QObject*, QEvent*) + 115
28  QtGui                         	0x0000000109a4d5fd QGuiApplicationPrivate::processExposeEvent(QWindowSystemInterfacePrivate::ExposeEvent*) + 317
29  QtGui                         	0x0000000109a493c5 QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) + 2165
30  QtGui                         	0x0000000109a36497 QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 215
31  QtGui                         	0x0000000109a34611 QWindowSystemInterface::flushWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 673
32  libqcocoa.dylib               	0x000000010bf993de 0x10bf89000 + 66526
33  QtGui                         	0x0000000109a53590 QWindow::setVisible(bool) + 912
34  QtWidgets                     	0x0000000109454e6b QWidgetPrivate::show_sys() + 827
35  QtWidgets                     	0x0000000109455daf QWidgetPrivate::show_helper() + 447
36  QtWidgets                     	0x0000000109456a42 QWidget::setVisible(bool) + 1394
37  QtWidgets                     	0x00000001095fcc4d QDialog::setVisible(bool) + 205
38  QtWidgets                     	0x00000001095fc293 QDialog::exec() + 307
39  com.impraxical.antimony       	0x0000000108febbeb App::onExportHeightmap() + 2795
40  QtCore                        	0x000000010a1e45f2 QMetaObject::activate(QObject*, int, int, void**) + 1986
[... cut ...]
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1067 MHz, 0x80CE, 0x4D34373142353237334348302D5946382020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xD1), Broadcom BCM43xx 1.0 (5.106.98.100.22)
Bluetooth: Version 4.2.7f4 15112, 3 services, 23 devices, 1 incoming serial ports
Network Service: AirPort, AirPort, en0
Serial ATA Device: APPLE SSD SM128C, 121,33 GB
USB Device: FaceTime Camera (Built-in)
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Apple Internal Keyboard / Trackpad
Thunderbolt Bus:

mereufrance77, 2015-02-18:
a try..

mkeeter, 2015-02-18:
Are you trying to save the image inside the `Antimony 0.7.5` mounted disk image?
This would explain the crash, as the disk image is read-only.
If so, try saving the image to a different directory (for example, saving to the Desktop)

astronouth7303, 2015-06-04:
@mereufrance77 Is this still a problem in 0.8.0?

mkeeter, 2015-09-16:
Closing in favor of #119, where we're making more progress.

closed 2015-09-16 by mkeeter