mkeeter/antimony #9: Crashes on MacbookAir

cashub, 2015-02-26:
Antimony (Release: 0.7.6b, feb 2015) crashes within a minute after launch and adding shapes (3D cube, sphere, cone). 

Hardware:
Macbook Air early 2014
OSX 10.10
CPU 1,4 GHz Intel Core i5
RAM 4 GB 1600 MHz DDR3
Graphics Intel HD Graphics 5000 1536 MB

mkeeter, 2015-02-26:
What does the crash log look like?  Is it something like this?
```
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff949c8286 __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff9850db53 abort + 129
2   libGPUSupportMercury.dylib    	0x00007fff96019bd9 gpusGenerateCrashLog + 173
3   com.apple.driver.AppleIntelHD4000GraphicsGLDriver	0x000012340035d5ed gpusKillClient + 9
4   libGPUSupportMercury.dylib    	0x00007fff9601b0c0 gpusSubmitDataBuffers + 500
5   com.apple.driver.AppleIntelHD4000GraphicsGLDriver	0x0000123400364030 IntelCommandBuffer::getNew(GLDContextRec*) + 48
6   com.apple.driver.AppleIntelHD4000GraphicsGLDriver	0x0000123400363ead intelSubmitCommands + 202
```

I've seen similar crashes on Yosemite machines with integrated GPUs; not yet sure of the cause.

andershaldin, 2015-04-01:
So you think it might be a hardware conflict? I'm running Ubuntu 14.04 on a Macbook Air and the last couple of weeks Antimony has been crashing really hard... to the point of where it is no longer usable :(

mkeeter, 2015-04-01:
@andershaldin, can you open up a new issue with details of your crashes?

I'm keeping this issue open for the Yosemite GPU crashes, but they may not related to crashes on Ubuntu (even on similar hardware).

joshuashort, 2015-05-29:
I saw this on Hackaday, and I had to rush to try it -- it really looks amazing!

Unfortunately, it crashes within a handful of seconds whenever it runs.. both the latest binary release and compiling from source. I'm trying it on a Yosemite Mac with Intel graphics. Here's the log..

```
Date/Time:       2015-05-29 12:06:32 -0700
OS Version:      10.10.3 (Build 14D136)
Architecture:    x86_64h
Report Version:  21

Command:         antimony
Path:            /Users/USER/Desktop/*/antimony.app/Contents/MacOS/antimony
Version:         ??? (0.0.0)
PID:             15783

Event:           spin-lite
Duration:        3.70s (process was unresponsive for 2 seconds before sampling)
Steps:           38 (100ms sampling interval)

Hardware model:  MacBookPro11,2
Active cpus:     8

Fan speed:       2158 rpm

Heaviest stack for the main thread of the target process:
  38  start + 52 (antimony + 19172) [0x10145fae4]
  38  main + 2178 (antimony + 397794) [0x1014bc1e2]
  38  QCoreApplication::exec() + 346 (QtCore + 2351962) [0x1027a635a]
  38  QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381 (QtCore + 2339709) [0x1027a337d]
  38  QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 2189 (libqcocoa.dylib + 131837) [0x1049ac2fd]
  38  -[NSApplication run] + 594 (AppKit + 551923) [0x7fff883efbf3]
  38  -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 346 (AppKit + 593768) [0x7fff883f9f68]
  38  _DPSNextEvent + 978 (AppKit + 596411) [0x7fff883fa9bb]
  38  _BlockUntilNextEventMatchingListInModeWithFilter + 71 (HIToolbox + 205099) [0x7fff8748712b]
  38  ReceiveNextEventCommon + 431 (HIToolbox + 205546) [0x7fff874872ea]
  38  RunCurrentEventLoopInMode + 235 (HIToolbox + 206191) [0x7fff8748756f]
  38  CFRunLoopRunSpecific + 296 (CoreFoundation + 465880) [0x7fff86d22bd8]
  38  __CFRunLoopRun + 927 (CoreFoundation + 467391) [0x7fff86d231bf]
  38  __CFRunLoopDoSources0 + 269 (CoreFoundation + 469901) [0x7fff86d23b8d]
  38  __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17 (CoreFoundation + 526849) [0x7fff86d31a01]
  38  QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) + 33 (libqcocoa.dylib + 136209) [0x1049ad411]
  38  QCocoaEventDispatcherPrivate::processPostedEvents() + 190 (libqcocoa.dylib + 134030) [0x1049acb8e]
  38  QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 971 (QtCore + 2353755) [0x1027a6a5b]
  38  QApplication::notify(QObject*, QEvent*) + 8136 (QtWidgets + 214600) [0x10192b648]
  38  QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251 (QtWidgets + 201515) [0x10192832b]
  38  QGraphicsScene::event(QEvent*) + 1451 (QtWidgets + 3463851) [0x101c44aab]
  38  QObject::event(QEvent*) + 755 (QtCore + 2528147) [0x1027d1393]
  38  QGraphicsScene::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) + 663 (QtWidgets + 3511447) [0x101c50497]
  38  QGraphicsScenePrivate::_q_emitUpdated() + 1682 (QtWidgets + 3410898) [0x101c37bd2]
  38  QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 971 (QtCore + 2353755) [0x1027a6a5b]
  38  QApplication::notify(QObject*, QEvent*) + 8136 (QtWidgets + 214600) [0x10192b648]
  38  QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251 (QtWidgets + 201515) [0x10192832b]
  38  QMainWindow::event(QEvent*) + 929 (QtWidgets + 1522465) [0x101a6ab21]
  38  QWidget::event(QEvent*) + 2836 (QtWidgets + 462548) [0x101967ed4]
  38  QWidgetPrivate::syncBackingStore() + 236 (QtWidgets + 397516) [0x1019580cc]
  38  QWidgetBackingStore::sync() + 499 (QtWidgets + 254131) [0x1019350b3]
  38  QWidgetBackingStore::doSync() + 2625 (QtWidgets + 251697) [0x101934731]
  38  QWidgetBackingStore::flush(QWidget*) + 371 (QtWidgets + 237555) [0x101930ff3]
  38  QWidgetBackingStore::qt_flush(QWidget*, QRegion const&, QBackingStore*, QWidget*, QPoint const&, QPlatformTextureList*, QWidgetBackingStore*) + 585 (QtWidgets + 236745) [0x101930cc9]
  38  QPlatformBackingStore::composeAndFlush(QWindow*, QRegion const&, QPoint const&, QPlatformTextureList*, QOpenGLContext*, bool) + 3523 (QtGui + 2734179) [0x10225a863]
  38  QOpenGLContext::swapBuffers(QSurface*) + 499 (QtGui + 379859) [0x10201bbd3]
  38  -[NSOpenGLContext flushBuffer] + 27 (AppKit + 3354427) [0x7fff8869bf3b]
  38  CGLFlushDrawable + 66 (OpenGL + 53178) [0x7fff88ef7fba]
  38  glSwap_Exec + 97 (GLEngine + 119089) [0x7fff8d3b0131]
  38  gldPresentFramebufferData + 142 (AppleIntelHD5000GraphicsGLDriver + 1085954) [0x123400109202]
  38  intelSubmitCommands + 191 (AppleIntelHD5000GraphicsGLDriver + 1086365) [0x12340010939d]
  38  IntelCommandBuffer::getNew(GLDContextRec*) + 48 (AppleIntelHD5000GraphicsGLDriver + 15512) [0x123400003c98]
  38  gpusSubmitDataBuffers + 142 (libGPUSupportMercury.dylib + 12034) [0x7fff8deb8f02]
  38  IOAccelContextSubmitDataBuffers + 245 (IOAccelerator + 6920) [0x7fff8f186b08]
  38  IOConnectCallStructMethod + 56 (IOKit + 27534) [0x7fff86c44b8e]
[... 15 lines cut ...]
 *38  lck_mtx_sleep + 134 (kernel + 1276342) [0xffffff80003379b6]
 *38  thread_block_reason + 175 (kernel + 1319343) [0xffffff80003421af]
 *38  ??? (kernel + 1329236) [0xffffff8000344854]
 *38  machine_switch_context + 367 (kernel + 2176511) [0xffffff80004135ff]

  Binary Images:
         0x10145b000 -        0x10154efff  com.impraxical.antimony (0.0.0)                                    <13D7F082-98F0-381A-AB9C-D7A4EB7E8B50>  /Users/USER/Desktop/*/antimony.app/Contents/MacOS/antimony
[... cut ...]
 *0xffffff7f8239c000 - 0xffffff7f823d6fff  com.apple.iokit.IOAcceleratorFamily2 156.14 (156.14)               <BE7D765B-49C1-34F9-B75E-3EAF8A4062A3>  /System/Library/Extensions/IOAcceleratorFamily2.kext/Contents/MacOS/IOAcceleratorFamily2
 *0xffffff7f82419000 - 0xffffff7f82466fff  com.apple.driver.AppleIntelHD5000Graphics 10.6.20 (10.0.6)         <F6346970-2B85-3F39-92FA-3CC28AEF9F89>  /System/Library/Extensions/AppleIntelHD5000Graphics.kext/Contents/MacOS/AppleIntelHD5000Graphics
 *0xffffff8000200000 - 0xffffff80009fffff  kernel (2782.20.48)                                                <4B3A11F4-77AA-3D27-A22D-81A1BC5B504D>  /System/Library/Kernels/kernel
 ```

joshuashort, 2015-05-29:
The graph view seems fine.. the hang/eventual crash happens when moving the mouse around the 'View'.

mkeeter, 2015-05-29:
Yeah, this is as-of-yet unsolved but very reproducible -- it happens whenever people try to use Yosemite with an integrated GPU.  I'm hoping that someone with this setup will do the deep device necessary to figure it out; I may also try to borrow a machine with that config to reproduce myself.

joshuashort, 2015-05-29:
Any hints? I have this setup, and I'm pretty familiar with Python and C++ .. I've used Qt a bit.. just not so much OpenGL..

dylanmc, 2015-05-29:
I have a MacBook Pro with Yosemite and both integrated and discrete GPU, and used the gfxCardStatus program to force my machine to stay with integrated graphics. Oddly, even in this mode, Antimony works fine (a bit slower). If I can run any experiments to help diagnose what's going wrong, I'm happy to.

codexmas, 2015-05-29:
Same issue here with Macbook Air, crashes after about 10 seconds. Also seeing a big lag between drags of objects and window positions before it finally crashes.

elbowdonkey, 2015-05-29:
Same issue here. I can include crash logs if that's useful.

Exact machine specs:

OSX: 10.10.3 (14D136)
GPU: Intel HD Graphics 5000 1536 MB
CPU: 1.7 GHz Intel Core i7
RAM: 8 GB 1600 MHz DDR3

I see the crash when simply switching from one window to another. When that happens, the first message I see in the logs is this: 

```
5/29/15 1:41:39.895 PM WindowServer[155]: Surface testing disallowed updates for 10 sequential attempts...
```

The resulting crash report has this:

```
Graphics hardware encountered an error and was reset: 0x00008807
```

elbowdonkey, 2015-05-29:
Is it possible that the version of OpenCL and OpenGL being used in antimony aren't supported on the Iris HD Graphics 5000 GPUs?

This Apple doc shows what's supported: https://support.apple.com/en-us/HT202823

My Macbook Air supports OpenGL 4.1 and OpenCL 1.2.

It's not obvious what versions are used by QT 5.4 - I'll continue to dig deeper.

mkeeter, 2015-05-29:
The crash is somewhere within Qt's redraw loop, so it's either Qt's fault or I'm doing something that corrupts OpenGL state.  If the crash is happening without any shapes present in the 3D view, then there's a pretty limited surface area:
https://github.com/mkeeter/antimony/blob/develop/src/ui/viewport/viewport.cpp#L488-L555
is the only chunk that messes with OpenGL in the 3D viewport.

I attempt to be very conservative with OpenGL, [targetting 2.1](https://github.com/mkeeter/antimony/blob/develop/src/app/main.cpp#L19-L20).

(I've also made plans to borrow a friend's machine with this configuration so I can reproduce myself).

joshuashort, 2015-05-29:
I dug around a little.. When I get back to my computer, I'll look more at the mouse move event and search for things regarding Qt5, OpenGL, Yosemite, and integrated graphics. Something's going on with mouse move.. if you move the cursor smoothly, slowly, and without much change in direction, it tends to work. Anything else will cause the app to hang and crash. Not all hangs cause a crash, but most do.

joshuashort, 2015-05-29:
Hmm.. I commented out lines 400 and 401 in viewport.cpp.. of course the coordinates don't refresh anymore, and it caused some insane graphics crash, but after that it seems to be working fine. I can actually interact with the view now, something that was impossible before taking out those lines.. panning, rotating, interacting with objects, it's working now..

elbowdonkey, 2015-05-30:
I fired the app up and avoided touching the trackpad. I used the keyboard to switch between windows (CMD  + ~) and there was no beach balling, no crashing whatsoever. I could have probably done that for hours.

Thinking that it might be a trackpad only thing I hooked up a USB mouse. I was able to switch between windows by clicking on their title bars, but it would get progressively slower and freeze the entire machine at times. Eventually the machine would become responsive again.

I then closed Antimony, and was about to write up this comment when I started seeing graphics artifacts in Safari, which was very quickly followed by a full kernel panic requiring a full power cycle.

There are some "mouse" related issues for QT. Maybe these are worth investigating? Though there's no smoking gun in any of these, and I'm not certain any of them are OpenGL related.

https://bugreports.qt.io/browse/QTBUG-44343
https://bugreports.qt.io/browse/QTBUG-39580
https://bugreports.qt.io/browse/QTBUG-39211

elbowdonkey, 2015-05-30:
As far OpenGL versions go, maybe 2.1 is too old? The mid-2007 machines support 3.3, while the latest machines support 4.1. I'll see if I can get a build with 3.3 working.

elbowdonkey, 2015-05-30:
I was able to build antimony using 3.3, but ran into the same issues.

I also just ran across this blog entry that might be relevant to the issues we're seeing: http://esoerik.blogspot.no/2015/01/in-pyqt5-with-qopenglwidget-on-os-x.html

elbowdonkey, 2015-05-30:
I think I found the issue. Can't say I know why because I found it accidentally.

If you disable multisampling by commenting out this line: https://github.com/mkeeter/antimony/blob/develop/src/app/main.cpp#L20 it works crash free.

lberezy, 2015-05-30:
Just chiming in to say I'm having the same issues with the current released binary.

My crash log is as below:

````
Process:               antimony [44807]
Path:                  /Volumes/VOLUME/antimony.app/Contents/MacOS/antimony
Identifier:            com.impraxical.antimony
Version:               0.7.9
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           antimony [44807]
User ID:               501

Date/Time:             2015-05-30 10:48:33.281 +1000
OS Version:            Mac OS X 10.10.3 (14D136)
Report Version:        11
Anonymous UUID:        171269B8-1F88-D142-D677-5389FEF1D116

Sleep/Wake UUID:       4AEF9FC7-409E-40B4-A7A4-52B947B05FC2

Time Awake Since Boot: 190000 seconds
Time Since Wake:       1300 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called

Application Specific Signatures:
Graphics hardware encountered an error and was reset: 0x00008807

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff85630286 __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff898f8b53 abort + 129
2   libGPUSupportMercury.dylib    	0x00007fff91367b81 gpusGenerateCrashLog + 173
3   com.apple.driver.AppleIntelHD5000GraphicsGLDriver	0x000012340044249f gpusKillClient + 9
4   libGPUSupportMercury.dylib    	0x00007fff91369068 gpusSubmitDataBuffers + 500
5   com.apple.driver.AppleIntelHD5000GraphicsGLDriver	0x0000123400003c98 IntelCommandBuffer::getNew(GLDContextRec*) + 48
6   com.apple.driver.AppleIntelHD5000GraphicsGLDriver	0x000012340010939d intelSubmitCommands + 191
7   QtWidgets                     	0x00000001025be9b8 0x10252e000 + 592312
8   QtWidgets                     	0x00000001025a367e QWidgetPrivate::sendComposeStatus(QWidget*, bool) + 46
9   QtWidgets                     	0x00000001025a36cc QWidgetPrivate::sendComposeStatus(QWidget*, bool) + 124
10  QtWidgets                     	0x00000001025a36cc QWidgetPrivate::sendComposeStatus(QWidget*, bool) + 124
11  QtWidgets                     	0x0000000102567c71 0x10252e000 + 236657
12  QtWidgets                     	0x0000000102567ff3 0x10252e000 + 237555
13  QtWidgets                     	0x000000010256b731 0x10252e000 + 251697
14  QtWidgets                     	0x000000010256c0b3 0x10252e000 + 254131
15  QtWidgets                     	0x000000010258f0cc QWidgetPrivate::syncBackingStore() + 236
16  QtWidgets                     	0x000000010259eed4 QWidget::event(QEvent*) + 2836
17  QtWidgets                     	0x00000001026a1b21 QMainWindow::event(QEvent*) + 929
18  QtWidgets                     	0x000000010255f32b QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
19  QtWidgets                     	0x0000000102562648 QApplication::notify(QObject*, QEvent*) + 8136
20  QtCore                        	0x00000001032f6a5b QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 971
21  QtWidgets                     	0x000000010286ebd2 0x10252e000 + 3410898
22  QtWidgets                     	0x0000000102887497 0x10252e000 + 3511447
23  QtCore                        	0x0000000103321393 QObject::event(QEvent*) + 755
24  QtWidgets                     	0x000000010287baab QGraphicsScene::event(QEvent*) + 1451
25  QtWidgets                     	0x000000010255f32b QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
26  QtWidgets                     	0x0000000102562648 QApplication::notify(QObject*, QEvent*) + 8136
27  QtCore                        	0x00000001032f6a5b QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 971
28  libqcocoa.dylib               	0x0000000105ef8b8e 0x105ed8000 + 134030
29  libqcocoa.dylib               	0x0000000105ef9411 0x105ed8000 + 136209
30  com.apple.CoreFoundation      	0x00007fff834eca01 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
31  com.apple.CoreFoundation      	0x00007fff834dec5c __CFRunLoopDoSources0 + 476
32  com.apple.CoreFoundation      	0x00007fff834de1bf __CFRunLoopRun + 927
33  com.apple.CoreFoundation      	0x00007fff834ddbd8 CFRunLoopRunSpecific + 296
[... 327 lines cut ...]
__GLSLBUILTINS                     2588K
__IMAGE                             528K
__LINKEDIT                         76.1M
__TEXT                            139.5M
__UNICODE                           552K
mapped file                        68.1M
shared memory                        68K
===========                      =======
TOTAL                             541.8M
TOTAL, minus reserved VM space    541.3M
 

Model: MacBookAir6,2, BootROM MBA61.0099.B18, 2 processors, Intel Core i5, 1.3 GHz, 4 GB, SMC 2.13f15
Graphics: Intel HD Graphics 5000, Intel HD Graphics 5000, Built-In
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E384A544D4C41522D4E544D
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E384A544D4C41522D4E544D
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x117), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.4f4 15601, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0128F, 121.33 GB
USB Device: Internal Memory Card Reader
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 23.6
````

elbowdonkey, 2015-05-30:
@lberezy are you able to compile from source? If so, I'd be interested to know if the fix I mentioned above works for you.

elbowdonkey, 2015-05-30:
@mkeeter I'm pretty confident this is the same issue we're seeing:

https://bugreports.qt.io/browse/QTBUG-46067

When I replace the format lines found in `main.cpp` with the following, it works, though the "View" window lacks a solid background. It's transparent, though everything else looks fine:

```
int main(int argc, char *argv[])
{
  QSurfaceFormat format;
  format.setDepthBufferSize(4);
  format.setMajorVersion(2);
  format.setMinorVersion(1);
  format.setSamples(2);
  format.setAlphaBufferSize(2);
  format.setProfile(QSurfaceFormat::CoreProfile);
  QSurfaceFormat::setDefaultFormat(format);
  
  // ... etc
  ```
I don't know anything about QT or its API, but I thought that setting an alpha buffer forces the alpha transparency on the window.  But if you remove that line, the app crashes the same way the current build crashes. In fact, if you omit any of the above lines, the crashes occur.

In any case, I think the solution to this issue lies somewhere in those QSurfaceFormat lines, as I've been able to run issue free when multisampling is turned off, or when I use the above code.

joshuashort, 2015-05-30:
Gotta hand it to ya, Michael, great find! I recompiled with the change you made, and it’s working.. very cool.

hgjfjfj, 2015-05-30:
Maybe related:

Process:               antimony [17347]
Path:                  /Volumes/VOLUME/antimony.app/Contents/MacOS/antimony
Identifier:            com.impraxical.antimony
Version:               0.7.9
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           antimony [17347]
User ID:               501

Date/Time:             2015-05-30 09:47:42.363 +0200
OS Version:            Mac OS X 10.10.3 (14D136)
Report Version:        11
Anonymous UUID:        9FE1B4EA-F168-6E8F-D069-482F165F3A9A

Sleep/Wake UUID:       03DC96EF-0098-4978-AF7C-1A63FD22ACC6

Time Awake Since Boot: 370000 seconds
Time Since Wake:       310000 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called

Application Specific Signatures:
Graphics hardware encountered an error and was reset: 0x00008807

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff9c3c5286 __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff9512db53 abort + 129
2   libGPUSupportMercury.dylib    	0x00007fff8ebc5b81 gpusGenerateCrashLog + 173
3   com.apple.driver.AppleIntelHD4000GraphicsGLDriver	0x0000123400369bc5 gpusKillClient + 9
4   libGPUSupportMercury.dylib    	0x00007fff8ebc7068 gpusSubmitDataBuffers + 500
5   com.apple.driver.AppleIntelHD4000GraphicsGLDriver	0x0000123400370640 IntelCommandBuffer::getNew(GLDContextRec*) + 48
6   com.apple.driver.AppleIntelHD4000GraphicsGLDriver	0x00001234003704bd intelSubmitCommands + 202
7   QtWidgets                     	0x000000010d8df9b8 0x10d84f000 + 592312
8   QtWidgets                     	0x000000010d8c467e QWidgetPrivate::sendComposeStatus(QWidget*, bool) + 46
9   QtWidgets                     	0x000000010d8c46cc QWidgetPrivate::sendComposeStatus(QWidget*, bool) + 124
10  QtWidgets                     	0x000000010d8c46cc QWidgetPrivate::sendComposeStatus(QWidget*, bool) + 124
11  QtWidgets                     	0x000000010d888c71 0x10d84f000 + 236657
12  QtWidgets                     	0x000000010d888ed1 0x10d84f000 + 237265
13  QtWidgets                     	0x000000010d88cd8a 0x10d84f000 + 253322
14  QtWidgets                     	0x000000010d88bcd6 0x10d84f000 + 249046
15  QtWidgets                     	0x000000010d8d9e58 0x10d84f000 + 568920
16  QtWidgets                     	0x000000010d88032b QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
17  QtWidgets                     	0x000000010d883648 QApplication::notify(QObject*, QEvent*) + 8136
18  QtCore                        	0x000000010e614d83 QCoreApplication::notifyInternal(QObject*, QEvent*) + 115
19  QtGui                         	0x000000010deb032d QGuiApplicationPrivate::processExposeEvent(QWindowSystemInterfacePrivate::ExposeEvent*) + 317
20  QtGui                         	0x000000010deac100 QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) + 2048
21  QtGui                         	0x000000010de992da QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 346
22  libqcocoa.dylib               	0x000000011117c411 0x11115b000 + 136209
23  com.apple.CoreFoundation      	0x00007fff9ad36a01 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
24  com.apple.CoreFoundation      	0x00007fff9ad28b8d __CFRunLoopDoSources0 + 269
25  com.apple.CoreFoundation      	0x00007fff9ad281bf __CFRunLoopRun + 927
26  com.apple.CoreFoundation      	0x00007fff9ad27bd8 CFRunLoopRunSpecific + 296
27  com.apple.HIToolbox           	0x00007fff911f956f RunCurrentEventLoopInMode + 235
28  com.apple.HIToolbox           	0x00007fff911f92ea ReceiveNextEventCommon + 431
29  com.apple.HIToolbox           	0x00007fff911f912b _BlockUntilNextEventMatchingListInModeWithFilter + 71
30  com.apple.AppKit              	0x00007fff959809bb _DPSNextEvent + 978
31  com.apple.AppKit              	0x00007fff9597ff68 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 346
32  com.apple.AppKit              	0x00007fff95975bf3 -[NSApplication run] + 594
33  libqcocoa.dylib               	0x000000011117b2fd 0x11115b000 + 131837
34  QtCore                        	0x000000010e61237d QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
35  QtCore                        	0x000000010e61535a QCoreApplication::exec() + 346
36  com.impraxical.antimony       	0x000000010d447286 main + 1718
[... 339 lines cut ...]
 

Model: MacBookAir5,2, BootROM MBA51.00EF.B00, 2 processors, Intel Core i5, 1.79 GHz, 10 GB, SMC 2.5f9
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Memory Module: BANK0/DIMM0, 8 GB, DDR3, 1333 MHz, Corsair, CMSO8GX3M1A1333C9
Memory Module: BANK1/DIMM1, 2 GB, DDR3, 1333 MHz, Elpida, EBJ20UF8BDU0-GN-0
AirPort: spairport_wireless_card_type_airport_extreme (0x168C, 0x8F), Atheros 9285: 4.0.74.0-P2P
Bluetooth: Version 4.3.4f4 15601, 3 services, 26 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
PCI Card: Broadcom Corporation NetLink BCM57766 Gigabit Ethernet PCIe, Ethernet Controller, PCI Slot 1
PCI Card: AR9285 Wireless Network Adapter, AirPort, PCI Slot 1
Serial ATA Device: WDC WD10SPCX-22HWST0, 1 TB
Serial ATA Device: ADATA SX300, 256,06 GB
Serial ATA Device: MATSHITADVD-RAM UJ8C2Q
USB Device: Hub
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: USB-PS/2 Optical Mouse
USB Device: SpaceNavigator
USB Device: Yubico Yubikey II
USB Device: Hub
USB Device: Bluetooth USB Host Controller
USB Device: HD WebCam
Thunderbolt Bus:

mkeeter, 2015-05-30:
Great detective work, folks.  I've removed the multisample line per @elbowdonkey's suggestion and pushed out a hotfix release ([0.7.9b](https://github.com/mkeeter/antimony/releases/tag/0.7.9b)).  Before I close this issue, can someone with the relevant hardware download it and test it?

mgold, 2015-05-30:
Sure, I stumbled on to the thread and can confirm 20 minutes or so of no crashing with 3D primitives and extruded shapes.

MacBook Pro (Retina, 13-inch, Early 2013)
Processor: 2.6 GHz Intel Core i5
Memory: 8 GB 1600 MHz DDR3
Graphics: Intel HD Graphics 4000 1024 MB

mkeeter, 2015-05-30:
Great!

closed 2015-05-30 by mkeeter