mkeeter/antimony #10: .stl file size.

andershaldin, 2015-03-04:
I get really huge file size when exporting to .stl, even for simple meshes. 
So obviously there is a balance between file size and quality of the mesh. But in this simple example I get the following file sizes

8 voxel = 3MB
16 voxel = 12.7MB
32 voxel = 51.7MB
64 voxel = 207.4MB

[image]
It seems a bit over the top to me, but not sure if there is anything we can do about it?

mkeeter, 2015-03-04:
Yes -- `.stl` export is a naive marching tetrahedrons algorithm at the moment (which quickly blows up in size).

Pull requests to improve it are welcome; for now, I'd bring the meshes into `meshlab` and decimate them there.

andershaldin, 2015-03-07:
Thanks for the reply, would love to help but my programming skill level is far to low :( 
I really like working in antimony but file size and export time is very unfortunate.  

Keep up the good work! :)

mkeeter, 2015-05-14:
The experimental feature detection in more recent builds lets you preserve sharp corners and edges while using a much lower voxel resolution.  It's not full-on decimation, but could help with filesize woes.

unlimitedbacon, 2015-06-26:
I think the thing that would help the most would be to merge groups of adjacent coplanar faces. Right now flat surfaces are composed of lots of small polygons, which are redundant.