mkeeter/antimony #12: Ubuntu 14.04 on a Macbook Air Crashes

andershaldin, 2015-04-06:
Open file (see download link) and rotate viewport. Consistant crach. I also got other files doing the same thing.

Terminal reads the following after crash:
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)

Hardware:
Prosessor: Intel® Core™ i7-3667U CPU @ 2.00GHz × 4 
Graphics: Intel® Ivybridge Mobile 
Memory:7.7 GiB

Ubuntu 14.04 LTS

Antimony file:
https://www.dropbox.com/s/xzyk1kcoa098h10/table.sb?dl=0

mkeeter, 2015-04-08:
It looks like that file is from a much older version of Antimony.  Can you update / rebuild Antimony (from the `develop` branch) and check if you're still seeing the frequent crashes?

I've upgraded your file so that it can be opened in more recent Antimony verisons and re-hosted it [here](http://tmp.cba.mit.edu/web/mkeeter/table.sb)

andershaldin, 2015-04-17:
My automatic update script did not work properly, am now up to date thanks! 
So many new nodes to play with! :)

closed 2015-04-17 by andershaldin