mkeeter/antimony #15: Arch Linux: cannot find -lboost_python-py34

Jarshvor, 2015-05-29:
Hi, thanks for this software mkeeter.
Its looking good! ;)

Wanted to comment that I just had to link some libraries to get it working on my Arch:

sudo ln -s /usr/lib/libboost_python3.so /usr/lib/libboost_python-py34.so

In case someone out there is having the same problem, make was spitting out:
/usr/bin/ld: cannot find -lboost_python-py34

Symlink fixes it. Thanks! :)

helb, 2015-05-29:
See the bottom of [BUILDING](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md) page – you could just change LIBS variable instead of creating symlinks in your system.

Jarshvor, 2015-05-29:
right!
damm, hadnt seen that.
thanks.
feel free to delete this thread?

mkeeter, 2015-05-29:
I've got an upcoming fix that will autodetect which library is present; leaving this issue open until it's committed.

C-o-r-E, 2015-05-29:
For reference, using the modification @helb was referring to does result in a build.

```-lboost_python3 ```

mkeeter, 2015-05-29:
f952d20b290c42e4c4cb65f29c126832be246c82 should fix this issue (it now checks for `libboost_python-py34` and `libboost_python3` and picks whichever one is present).

closed 2015-05-29 by mkeeter