mkeeter/antimony #16: Linux Mint 16 - cannot find -lboost_python-py34 and -lpython3.4m

zaidpirwani, 2015-05-29:
says this...

r.o home/zaid/antimony/build/moc_depth_image.o home/zaid/antimony/build/moc_editor.o home/zaid/antimony/build/moc_script_pane.o home/zaid/antimony/build/moc_button.o home/zaid/antimony/build/moc_resolution_dialog.o home/zaid/antimony/build/moc_render_task.o home/zaid/antimony/build/moc_render_worker.o home/zaid/antimony/build/moc_render_image.o home/zaid/antimony/build/moc_export_mesh.o home/zaid/antimony/build/moc_export_worker.o home/zaid/antimony/build/moc_export_heightmap.o   -lpng -lboost_python-py34 -lpython3.4m -L/home/zaid/Qt5.4.0/5.4/gcc_64/lib -lQt5OpenGL -lQt5Widgets -lQt5Gui -lQt5Network -lQt5Core -lGL -lpthread 
mkdir -p /home/zaid/antimony/build/sb
cp -f -R /home/zaid/antimony/qt/../py/nodes /home/zaid/antimony/build/sb
cp -f -R /home/zaid/antimony/qt/../py/fab /home/zaid/antimony/build/sb
/usr/bin/ld: cannot find -lboost_python-py34
/usr/bin/ld: cannot find -lpython3.4m
collect2: error: ld returned 1 exit status
make: *** [antimony] Error 1
zaid@zp:~/antimony/build$

mkeeter, 2015-05-29:
Have you tried the last suggestion in the [Caveats](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md#caveats) list?

On some systems (e.g. Fedora 21), the name for `boost::python` is different.

Change [this line](https://github.com/mkeeter/antimony/blob/develop/qt/shared.pri#L44) to read
```
LIBS += -lboost_python3
```
(or whatever the library is named).

zaidpirwani, 2015-05-29:
yes, I did try replacing that line... still didn't work.

it says
cannot find -lboost_python3

mkeeter, 2015-05-29:
Can you look for the library in your linker path?
Try running ` ldconfig -p | grep libboost_python`

astronouth7303, 2015-06-04:
@zaidpirwani Did you ever solve this? Does the latest version compile?

zaidpirwani, 2015-06-05:
yeah I had got it to work somehow... forgot how..

let me try again, removing and trying again.

zaidpirwani, 2015-06-05:
Still, I get this

zaid@zp:~/Desktop$ git clone https://github.com/mkeeter/antimony
Cloning into 'antimony'...
remote: Counting objects: 16134, done.
remote: Compressing objects: 100% (62/62), done.
remote: Total 16134 (delta 21), reused 0 (delta 0), pack-reused 16072
Receiving objects: 100% (16134/16134), 3.07 MiB | 444.00 KiB/s, done.
Resolving deltas: 100% (11673/11673), done.
Checking connectivity... done.
zaid@zp:~/Desktop$ cd antimony
zaid@zp:~/Desktop/antimony$ mkdir build
zaid@zp:~/Desktop/antimony$ cd build
zaid@zp:~/Desktop/antimony/build$ 
zaid@zp:~/Desktop/antimony/build$ ~/Qt5.4.0/5.4/gcc_64/bin/qmake ../qt/antimony.pro
Project ERROR: Could not find boost::python3
zaid@zp:~/Desktop/antimony/build$ make -j8
make: *** No targets specified and no makefile found.  Stop.
zaid@zp:~/Desktop/antimony/build$ 

These commands were run after installing all the dependencies as provided in the BUILDING.md file

However, I remember that last time, I opened it in qt and compiled it from there...

Posting steps.

zaidpirwani, 2015-06-05:
[image]
Opening in Qt Creator

[image]
Configuring Project

[image]
ERROR

Removed the NEW stuff in the shared.pri file - inside the Linux section with this
linux {
    QMAKE_CXXFLAGS += $$system(/usr/bin/python3-config --includes)
    QMAKE_LFLAGS   += $$system(/usr/bin/python3-config --ldflags)
    LIBS += -lboost_python-py33 -lpython3.3m
}

it shows the DEBUG button now working in Qt and I can hit Debug/Run/Play [whichever it is]
It WORKs..!
[image]

I am sorry I am not much comfortable with the scripts you have added, I can somewhat understand em, but i don't know why the issue is arising.

mkeeter, 2015-06-05:
Okay, that sounds similar to #18 (special-casing for systems where you have Python 3.3 rather than 3.4).
Glad that things are working for you!

closed 2015-06-05 by mkeeter