mkeeter/antimony #17: License is unclear

hhhudson, 2015-05-29:
Neat project.

Would it possible to consider transitioning the project to an established licensing scheme, such as the MIT license? The current license is very vague and technically forbids commercial use (as it is not explicitly permitted), although I'm not sure if that was the intention. This makes the software non-free by free/oss definitions.

clothbot, 2015-05-29:
+1 for clarifying the FOSS-compatible licensing options.  Looking forward to playing+contributing to this project!

carter2422, 2015-05-29:
I'm also interested in a more clear license :+1:

mkeeter, 2015-05-30:
Fixed (a519b935749a52feba3dfba06bdb4659a35b8103).

closed 2015-05-30 by mkeeter

clothbot, 2015-05-30:
Excellent!  Thanks for explicitly stating that it's the MIT license.