mkeeter/antimony #19: Separate Node editor into a separate project

prusnak, 2015-05-29:
Wow! Antimony looks great!

Some time ago I wanted to write my own piece of software inspired by Max/MSP or VVVV and started to design my own node editor. After few days I abandoned the project because of lack of the time.

I would love to see the node editor extracted from the current codebase into a separate project which would be working on abstracted classes and thus would be usable for writing lots of other visual programming tools for creating sounds, music, graphics, videos, controlling hardware, etc.

astronouth7303, 2015-06-04:
@mkeeter How do you feel about this?

mkeeter, 2015-06-05:
I'll tag this as "dreams" -- interesting aspirational task, but unlikely to get done in the near future.

joshuashort, 2015-06-05:
I like the idea -- a good graph/node editor is a handy thing to have. Certainly, there are plenty of other things to focus on within the project; it would some complexity (especially maintaining it as a separate project). I don't have a good sense of how tight the coupling is.. but it'd be a nice goal to have the canvas as separable as reasonably possible.

mkeeter, 2015-07-31:
I'm counting this as done with the new graph engine (which can be built as a stand-alone library, has its own unit test suite, etc).

closed 2015-07-31 by mkeeter