mkeeter/antimony #20: Segfault while mouse-moving the center of a polar array

Zopieux, 2015-05-29:
`Linux 4.0.4-2-ARCH #1 SMP PREEMPT Fri May 22 03:05:23 UTC 2015 x86_64 GNU/Linux`

```text
Core was generated by `./antimony'.
Program terminated with signal SIGSEGV, Segmentation fault.
(gdb) bt
#0  0x00007f9bd79fb79a in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) ()
   from /usr/lib/libQt5Core.so.5
#1  0x00007f9bd8a2c3bc in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /usr/lib/libQt5Widgets.so.5
#2  0x00007f9bd8a31820 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#3  0x00007f9bd79fb9eb in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/libQt5Core.so.5
#4  0x00007f9bd79fda2b in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) ()
   from /usr/lib/libQt5Core.so.5
#5  0x00007f9bd7a539f3 in ?? () from /usr/lib/libQt5Core.so.5
#6  0x00007f9bd5f619fd in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#7  0x00007f9bd5f61ce0 in ?? () from /usr/lib/libglib-2.0.so.0
#8  0x00007f9bd5f61d8c in g_main_context_iteration () from /usr/lib/libglib-2.0.so.0
#9  0x00007f9bd7a53de7 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/libQt5Core.so.5
#10 0x00007f9bd79f9482 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/libQt5Core.so.5
#11 0x00007f9bd7a00e5c in QCoreApplication::exec() () from /usr/lib/libQt5Core.so.5
#12 0x000000000046085f in main ()
```

astronouth7303, 2015-06-04:
Works for me. Can you give more detail what you were doing?

Zopieux, 2015-06-05:
Pretty hard to describe, it seems pretty random (see #29). I created a few objects (circle, extrude, other circle, other extrude, polar array of these, difference the extrude and the extrude array) and then moved the array center with the mouse in the preview buffer. A few pixels in, boom, headshot.

mkeeter, 2015-06-05:
Can you make a debug build so that we get a useful stack trace out?
Instead of the normal `qmake` invokation, use `qmake CONFIG+=debug ...`, then run in `gdb` or `lldb` and see where things break.