mkeeter/antimony #21: Antimony fails to start due to import error

C-o-r-E, 2015-05-29:
Upon trying to start antimony, a dialog is presented with the text: 
``` 
Import Error:

Could not find fab Python module. 
Antimony will now exit. 
```

There is no trace presented on the command line and the program exits.

Build from latest develop branch on Archlinux with Qt 5.4.1. Only modification was to the line modification specified in the BUILDING file.

``` LIBS += -lboost_python3 ```

mkeeter, 2015-05-29:
Are you running it from the build directory, or did you run `make install`?

Normally, Antimony will produce a folder structure `sb/fab/...` that contains the standard shapes library and a few magic hooks.  Do you see that folder?

C-o-r-E, 2015-05-29:
Running from the build directory

```
$ ls sb/
__init__.py  __pycache__  fab  nodes  shapes.py  types.py  ui.py

$ ls sb/fab/
__init__.py  __pycache__  shapes.py  types.py  ui.py
```

mkeeter, 2015-05-30:
7ef4451f1168a78a65a3d46f7899cc9ef0aae5d6 will print out the Python error on the console.  Can you pull / rebuild and see what it says?

C-o-r-E, 2015-05-30:
Now getting a traceback:

```
$ ./antimony 
Traceback (most recent call last):
  File "/home/corey/git/antimony/build/sb/fab/__init__.py", line 1, in <module>
    import fab.shapes as shapes
  File "/home/corey/git/antimony/build/sb/fab/shapes.py", line 2, in <module>
    import functools
  File "/usr/lib/python3.4/functools.py", line 22, in <module>
    from types import MappingProxyType
ImportError: cannot import name 'MappingProxyType'
```

mkeeter, 2015-05-30:
It looks like there's a conflict between `fab.types` and the system's `types` module.  I'm able to reproduce this by running `python3` from the `sb/fab` folder then trying `import functools`, but not sure why it's happening when you run Antimony from the build directory.

Silly question -- do you see `types.py` in the build directory (from where you're running Antimony)?  If you just run `python3 -c "import functools"` from the folder from where you're running Antimony, does it fail in the same way?

C-o-r-E, 2015-05-30:
I dont seem to have types.py in the build directory...

```
$ pwd
/home/corey/git/antimony/build
$ ls
Makefile                    moc_int_datum.cpp
antimony                    moc_link.cpp
home                        moc_main_window.cpp
moc_app.cpp                 moc_name_datum.cpp
moc_button.cpp              moc_node.cpp
moc_canvas.cpp              moc_output_datum.cpp
moc_connection.cpp          moc_port.cpp
moc_control.cpp             moc_proxy.cpp
moc_control_root.cpp        moc_render_image.cpp
moc_datum.cpp               moc_render_task.cpp
moc_depth_image.cpp         moc_render_worker.cpp
moc_deserializer.cpp        moc_resolution_dialog.cpp
moc_editor.cpp              moc_root.cpp
moc_eval_datum.cpp          moc_script_pane.cpp
moc_export_heightmap.cpp    moc_serializer.cpp
moc_export_mesh.cpp         moc_shape_datum.cpp
moc_export_worker.cpp       moc_shape_output_datum.cpp
moc_float_datum.cpp         moc_view_selector.cpp
moc_float_output_datum.cpp  moc_viewport.cpp
moc_graph_scene.cpp         moc_viewport_scene.cpp
moc_input.cpp               qrc_gl.cpp
moc_inspector.cpp           sb
moc_inspector_buttons.cpp   ui_exporting_dialog.h
moc_inspector_row.cpp       ui_main_window.h
moc_inspector_text.cpp      ui_resolution_dialog.h
moc_inspector_title.cpp
```

Importing functools returns no error.

C-o-r-E, 2015-06-01:
It seems to work now.
I have since deleted the build directory and started from scratch. Perhaps there was a problem with the initial invocation of qmake? 

Closing the issue for now

closed 2015-06-01 by C-o-r-E