mkeeter/antimony #24: is it possible to refer to another node by reference?

jschachter, 2015-05-29:
I want to refer to a connected node's parameters. For example in the "iterate (2d)" function, 'i.r' isn't gets an error (i is not defined.)

mkeeter, 2015-05-29:
I'm not entirely sure what you're asking, but you can refer to nodes by reference.

Every node has a unique name (shown in the top-left corner of its box in the graph).  You can click on this box to edit the name, then refer to properties in the form `node_name.prop_name`.

For example, if you have a point named `p`, you could use the expression `p.x` in other scripts.

jschachter, 2015-05-29:
No, those are by name. How do I refer to a parent (or child's) parameter without knowing its name? Say you want to a extrude a circle by twice its radius. You could just connect to the circle's radius...

mkeeter, 2015-05-29:
Ah, I see.

Yes, in that case the easiest way to do it would be to edit the `Extrude` node, giving it another `float` input port that you plug the radius into (and then using that value in the script).

e.g.

```python
import fab

input('zmin', float)
input('r', float)
input('shape', fab.types.Shape)

output('out', fab.shapes.extrude_z(input, zmin, zmin + 2*r))
```

closed 2015-06-04 by astronouth7303