mkeeter/antimony #25: Data Provider Node Type

mikea, 2015-05-30:
A separate "data provider" node type seems to be useful. I imagine it could have multiple (named) rows of values. These would be used as providers for common parameters through a model (that doesn't have to be changed by mouse).

mkeeter, 2015-05-30:
You can do this easily with a custom script, e.g.

```py
title('data provider')
input('width', float)
input('height', float)
input('depth', float)
```

Name the node `constants`, then refer to `constants.width`, etc. in other places.

[image]

Is this what you are asking?

astronouth7303, 2015-06-04:
[I just wrote one](https://groups.google.com/d/topic/antimony-dev/jjG4v9Hj7FE/discussion).