mkeeter/antimony #26: Functions for variable connection.

mikea, 2015-05-30:
To keep things centered in the model I'd like to specify a function for a wire (e.g. x / 2)

astronouth7303, 2015-05-30:
+1

I was doing the screwdriver demo and I wanted this for the inserts (eg, extrude from -radius to +xmax).

mkeeter, 2015-05-30:
Interesting.  What do you see this looking like in the UI?  I could imagine a smaller text box midway through the connection with a single expression that's evaluated (maybe using `_` as the input variable).

astronouth7303, 2015-05-30:
Maybe just a node with a number input (x, a) and a number output (y, b), but no shapes?

mkeeter, 2015-05-30:
That's already pretty easy with a custom script, something like
```py
title("/2")
input("a", float)
output("b", a/2)
```
Does that do what you're asking?

Workflows vary from person to person, but when designing anything interesting, I end up spending a lot of time customizing the pre-existing nodes by editing their scripts (rather than trying to find ones that do *exactly* what I need and connecting them).

astronouth7303, 2015-05-30:
Yeah, that works. But it'd be nice if I could have a node like

     |         |
----|-------
a | `2`
b | `-a`

So that simple one-line expressions don't involve writing a full node.

Salandora, 2015-05-31:
I solved it this way:

    title("Multiply")
    input("a", float)
    input("mul", float)

    output("b", a*mul)
this allows me to set mul to -1 for b = -a
and 0.5 for b = a/2

mgold, 2015-05-31:
I found myself wanting this too. Treat constants, not just shapes, as data in the DAG.

unlimitedbacon, 2015-06-02:
This is how I imagine this should be handled in the UI:

[image]

Expressions like that are already evaluated correctly. All thats needed is to draw the connection between the nodes and highlight the names of the variables.

Conceivably, this would also allow you to have multiple outputs feeding into the same input. These would just be represented as different variables in the expression.

astronouth7303, 2015-06-02:
Hm. Making node reference more prominent would certainly solve the problems described here, and drawing lines would make it more obvious of the connection. 

(As a nitpick, maybe draw them in a different style (thinner, dashed, gray) so that it's obvious to the user that this is an indirect link and non-interactive.)