mkeeter/antimony #27: Attribute Error in Function node

astronouth7303, 2015-05-30:
Compiled from git version 27fe5f6 (master, 0.7.9).

Function (prefix, xy):
```
Traceback (most recent call last):
  File "<string>", line 34, in <module>
AttributeError: 'module' object has no attribute 'function_prefix_xy'
```

Function (prefix, xyz):
```
Traceback (most recent call last):
  File "<string>", line 34, in <module>
AttributeError: 'module' object has no attribute 'function_prefix_xy'
```

mkeeter, 2015-05-30:
This looks like a mistmatch between `shapes.py` and the nodes library (especially if you had previously built from the `develop` branch).  Can you try deleting the `sb` folder and re-building to make sure that everything is copied over correctly?

As an aside, I'd recommend running on the `develop` branch -- I try to keep it pretty stable, and it's usually a few weeks ahead of the public releases on `master`.

astronouth7303, 2015-05-30:
Huh, why doesn't `make clean` get rid of that? (And while I'm at it, it doesn't actually remove the built binary. And there's a strange folder called `home`.)

Not sure if that fixed it, though, because now I'm getting reference errors on compile.

astronouth7303, 2015-05-30:
Yeah, getting a ton of link errors on `develop`. However, it seems to be fixed now when I compile `master`.

closed 2015-05-30 by astronouth7303