mkeeter/antimony #29: Segmentation Fault while Rotating 3D view

der-coder, 2015-05-30:
gdb output:

    gdb ./antimony
    Reading symbols from ./antimony...(no debugging symbols found)...done.
    (gdb) run
    Starting program: /home/isaac/Programs/antimony-build/antimony 
    [Thread debugging using libthread_db enabled]
    Using host libthread_db library "/usr/lib/libthread_db.so.1".
    [New Thread 0x7fffd2785700 (LWP 3821)]
    [New Thread 0x7fffd2f86700 (LWP 3820)]
    [New Thread 0x7fffdda06700 (LWP 2787)]
    [New Thread 0x7fffe9eae700 (LWP 2786)]
    Program received signal SIGSEGV, Segmentation fault.
    0x00007ffff572b79a in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () from /usr/lib/libQt5Core.so.5

       (gdb) bt
    #0  0x00007ffff572b79a in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*,     QEvent*) () from /usr/lib/libQt5Core.so.5
    #1  0x00007ffff675c3bc in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/lib/libQt5Widgets.so.5
    #2  0x00007ffff6761820 in QApplication::notify(QObject*, QEvent*) ()
       from /usr/lib/libQt5Widgets.so.5
    #3  0x00007ffff572b9eb in QCoreApplication::notifyInternal(QObject*, QEvent*)
        () from /usr/lib/libQt5Core.so.5
    #4  0x00007ffff572da2b in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /usr/lib/libQt5Core.so.5
    #5  0x00007ffff57839f3 in ?? () from /usr/lib/libQt5Core.so.5
    #6  0x00007ffff3c919fd in g_main_context_dispatch ()
       from /usr/lib/libglib-2.0.so.0
    #7  0x00007ffff3c91ce0 in ?? () from /usr/lib/libglib-2.0.so.0
    #8  0x00007ffff3c91d8c in g_main_context_iteration ()
       from /usr/lib/libglib-2.0.so.0
    #9  0x00007ffff5783de7 in     QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
    #10 0x00007ffff5729482 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
    #11 0x00007ffff5730e5c in QCoreApplication::exec() ()
       from /usr/lib/libQt5Core.so.5
    #12 0x000000000046085f in main ()
    (gdb) 

My system info:

    OS: Arch Linux 
    Kernel: x86_64 Linux 4.0.4-2-ARCH
    CPU: Intel Core i5-3337U CPU @ 2.7GHz

Steps to reproduce:
1. Create a Circle of radius 15
2. Create a Rectangle of Height 20 and Length 30
3. Extrude both objects
4. Zoom out and Rotate the 3D view until it gets a Segmentation error

antimony build flags
    make -j4 -o3

Zopieux, 2015-05-30:
Seems related to my segfault as in #20.

astronouth7303, 2015-06-04:
I'm unable to reproduce.