mkeeter/antimony #30: suggestion: change default Add menu keybinding

xxv, 2015-05-30:
`Shift+A` is good, but not great. Perhaps a cue could be borrowed from Blender and the spacebar could bring up the add menu? right-click (I know, many mac users)? I find it a bit tedious to add a bunch of things and faster access to add could help a ton.

joshuashort, 2015-05-30:
I assumed right-click would work.. even Mac users are ok with this ;)
It seems natural for a 'right-click' or 'two-finger click' to bring up a menu..

astronouth7303, 2015-05-30:
+1 Right click

joshuashort, 2015-05-31:
I forked and modified viewport.cpp and canvas.cpp to show the add menu when right clicking; it's basically copy-pasta from the keyPressEvent; it'd probably be best to create some kind of show-menu-helper-function, but this does work for now. I'll submit a pull request, if you want it. I haven't done this before, so hopefully it goes well.

mgold, 2015-05-31:
+1 Right click (Mac user)

astronouth7303, 2015-06-04:
Fixed by #42

closed 2015-06-04 by astronouth7303