mkeeter/antimony #31: rounded rectangle construction

xxv, 2015-05-30:
Doing an Offset transform on a rounded rectangle reveals the construction of the rounded rectangle, which is a non-intuitive result.

mkeeter, 2015-05-30:
This is because the offset function is a bit of a hack: it just slides up and down the distance field, and Antimony's distance fields are
- Not necessarily Euclidean
- Chosen to be right shape at the zero crossing (but not necessarily everywhere else)

Here's the distance field for a rounded rect:
[image]

To make offset well-behaved in all cases, we'd actually have to take level sets of a distance transform on the zero-crossing of the original field.  Could be a fun blog post / conference paper -- any volunteers?

astronouth7303, 2015-05-30:
We're just talking 2D paths, right? How does SVG/Inkscape do it?

mkeeter, 2015-05-30:
It's a bit trickier than that.  The rounded rectangle is represented internally by a function `f(x,y,z)` that returns a floating-point number.  If `f(x,y,z) < 0`, then we're inside the shape; otherwise, we're outside.  The edge of the rectangle is the set of points where `f(x,y,z) = 0`; the boundary is represented implicitly, rather than explicitely (as in SVGs and other boundary representations).

Doing CAD with implicit surfaces makes writing a geometry / CSG kernel very easy (unioning two shapes is just `min(a(x,y,z), b(x,y,z))`), but operations on specific features (like the edge of the rectangle) become harder to represent.

Further reading:
- [Functional representation](http://en.wikipedia.org/wiki/Function_representation)
- [Implicit surface](http://en.wikipedia.org/wiki/Implicit_surface)
- [Boundary representation](http://en.wikipedia.org/wiki/Boundary_representation)
- [My thesis](mattkeeter.com/research/thesis.pdf) (PDF)

astronouth7303, 2015-05-30:
Ah, hence the problems with interacting with other formats and difficulties with "free drawing" methods.