mkeeter/antimony #36: Add menu acellerators & keyboard shortcuts

virtualritz, 2015-05-30:
See title.

mkeeter, 2015-05-30:
Have you found `Shift + A`?

xxv, 2015-05-30:
`Shift + A` helps, but I think Antimony could really benefit from some way to quickly drill down once you have it opened. Accelerators are good and easy to add, but I've found that pop-up substring searching (like Blender) is one of the most powerful tools. It'd be even more powerful if you were to somehow display iconic forms instead of just words to help someone find a desired form quickly.

joshuashort, 2015-05-31:
While it's undoubtedly a lot of nice-to-have work, it'd be nice to have the ability to define keyboard shortcuts for some actions.. right now, the add menu seems to be the main one.

astronouth7303, 2015-06-04:
Given that the set of nodes is determined at run time (and, imho, should vary from user to user), this would be hard to have for nodes. A macro system would work better for this.

What other actions are missing keyboard shortcuts?

mkeeter, 2015-06-05:
I'm going to reframe this as a blender-style quick-search tool (as @xxv mentioned) -- spacebar should open up a search menu does substring matching against node names in your library.

joshuashort, 2015-06-05:
I like the idea of being able to define macros.. especially when combined with the ability to make node assemblies. The quick-search idea is also very cool.

As far as missing keyboard shortcuts.. what comes to mind is coupled with actual missing functionality.. unless I'm wrong, we can't currently select all nodes from the viewport? And this is getting a bit off topic, but the ability to select all visible nodes might also be handy?

astronouth7303, 2015-06-05:
I've also wondered about a standard group select.

benwhalley, 2018-12-08:
Sublime text has a nice thing going with cmd-P, cmd+shift+p and cmd-R... Perhaps worth looking at editors for nice ways to expose lots of context-sensitive functions?