mkeeter/antimony #37: Add chamfer transform

xxv, 2015-05-30:
I have no idea how hard this would be considering the data model, but it'd be very handy to be able to chamfer the corners of a 2D input shape. I imagine the inputs would be:

* offset
* shape

astronouth7303, 2015-05-30:
I'm also curious as to how to do this. Because of the functional representation (#31), I expect this is actually pretty difficult.

astronouth7303, 2015-06-02:
I have two thoughts on how to to implement this:

1. Have a series of chamfered/rounded primitives that can be used to build up an object with edges. Not easy to use, and hard to add to an object later.

2. Have "routing" operations (as in, with a router bit) that do this. Things like line, arc, rectangle, circle, etc. Or: Routing primitives meant to be subtracted from an object.

I like #2 better.

astronouth7303, 2015-06-04:
Ok, I've got a basic [90deg straight-line fillet node](https://gist.github.com/astronouth7303/c6c1fa2065b63b7e3352). Use Difference for convex corners, Union for concave corners.

It will trace the line from _(x0, y0, z0)_ to _(x1, y1, z1)_ and set a round of radius _r_ (in model units). Use _a_ to adjust the angle. (_a_ is a little opaque, sorry. Will be improved in future versions.)

I highly suggest binding it to your model.

astronouth7303, 2015-06-04:
Future plans:

* More primitive shapes, namely interior and exterior arcs
* Versions of everything in chamfer
* Arbitrary joining angles