mkeeter/antimony #38: Freaking Awesome

ylluminate, 2015-05-30:
Hey, just wanted to give you a quick shoutout (pardon my abuse of github issues) about how awesome this is.  I think this is the most well thought out CAD system I've laid my hands on.  I love the node based approach and therefore quick ability to see the entire system together and how it is connected.  Great job!

mgold, 2015-05-30:
You know, maybe this means it's time for a mailing list?

Also a roadmap. If I were to contribute, I'd want to do something I knew was useful and would be merged.

dylanmc, 2015-05-30:
If there's a freaking awesome thread, I'm all over that one. +1 here.

mkeeter, 2015-06-01:
Alright, I've created a [mailing list](https://groups.google.com/forum/#!forum/antimony-dev).  The [next release](https://github.com/mkeeter/antimony/tree/release/0.8.0) focuses on polishing the standard libraries and a handful of minor tasks; once it goes out, I'll take some time to think about a bigger roadmap (suggestions welcome!).

jasonm23, 2015-06-01:
:+1: +1 absolutely great!

ccorcos, 2015-06-02:
I was just thinking how awesome this is. I've been using OpenSCAD in some side projects but a GUI / interactive renderer is so much more convenient for certain things. Very good job with this.

Also, I'd love to hear more information about how this software was built. I've always wanted to learn how to build a polished desktop application like this. I see you're using qt and python. I'd appreciate a simple walkthrough of how this project is laid out so I can learn how it works and hopefully contribute.

closed 2015-06-04 by astronouth7303

mkeeter, 2015-06-04:
@ccorcos I've sent out a brief code map on the [antimony-dev](https://groups.google.com/forum/#!topic/antimony-dev/3Hbfa3TKyc0) mailing list -- it should at least give you a frame of reference in exploring the source.