mkeeter/antimony #39: Linking error in develop

astronouth7303, 2015-05-30:
It seems that it's no longer linking in boost in d2ad474 on my system. ([Partial build log](http://paste2.org/x76m3Ep4))

My build process is:
```
$ make clean
$ rm -r *
$ ~/.local/Qt5.4.1/5.4/gcc_64/bin/qmake ../qt/antimony.pro && make 
```

I've made no changes to the project.

mkeeter, 2015-05-31:
I suspect that f952d20b290c42e4c4cb65f29c126832be246c82 is the culprit.  Can you run `ldconfig -p|grep python` and see what boost::python is named?  Right now, I check for `libboost_python-py34.so` and `libboost_python3.so`, using whichever one is present.

astronouth7303, 2015-05-31:
Ah, that would be the problem. For whatever reason, `ldconfig` is under `/sbin`, which is only on the path when you're root.

I'm not certain if this is a bug with Debian testing or a misuses of `ldconfig` on the part of `qmake`.

astronouth7303, 2015-05-31:
So `antimony.pro` uses `ldconfig` to detect how `boost::python` is installed. (This appears to be a Debian/Ubuntu vs Arch & Fedora thing.)

Arch installs `ldconfig` in `/usr/bin`. Ubuntu and Debian install it in `/sbin`. Fedora has it in `/sbin`.

mkeeter, 2015-05-31:
This looks like it's an issue with the default `$PATH` variable.  It works fine with Ubuntu because `/sbin` is [in the default path](http://askubuntu.com/questions/386629/what-are-the-default-path-values).

Is Debian testing doing something different with `$PATH`?

astronouth7303, 2015-05-31:
Debian only has `/sbin` in the PATH if you're root.

astronouth7303, 2015-05-31:
Crowdsourced answer for Fedora: `/sbin` is on the path by default.

mkeeter, 2015-05-31:
Okay, want to try ace33e00a62393b7cc157e836875aebcaba55cf5?  The Linux build is getting a bit hairy, but it should now look for `ldconfig` if it's not on the path by default.

astronouth7303, 2015-05-31:
(For completeness, Arch also has `/sbin` on the PATH.)

astronouth7303, 2015-05-31:
Yup, builds now. Thank you!

(I think this is why GNU configure generally takes the approach of "build a bunch of test programs and see which ones compile".)

closed 2015-05-31 by astronouth7303