mkeeter/antimony #46: missing lib on ubuntu

oli44, 2015-06-01:
Hello 

running linux mint 17 Qiana. When trying to compile after succesfully installing all dependencies, trouble starts:

    $ qmake ../qt/antimony.pro
    sh: 1: /usr/bin/python3-config: not found
    sh: 1: /usr/bin/python3-config: not found
    sh: 1: /usr/bin/python3-config: not found

    $ whereis qmake
    qmake: /usr/bin/qmake /usr/bin/X11/qmake

Any idea?

mkeeter, 2015-06-01:
Can you look for `python3-config`, e.g. with `which python3-config` (or `whereis python3-config`)?

oli44, 2015-06-01:
Not much to read:

     $ which python3-config 
     $ whereis python3-config 
         python3-config

     $ apt-cache search python3-config
     python3-configglue - Rassemble optparse.OptionParser et ConfigParser.Config.Parser

    $ uname -a
    Linux xmg 3.13.0-53-generic #89-Ubuntu SMP Wed May 20 10:34:39 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

mkeeter, 2015-06-01:
Are you sure you've installed `python3-dev`?

(`sudo apt-get install python3-dev`)

oli44, 2015-12-07:
Hello,

back on antinomy. 0.90c.
In the meanwhile I did install python3-dev.

bug solved, we may close it

closed 2015-12-07 by oli44