mkeeter/antimony #50: Suggestion: Easier node connection

Highstaker, 2015-06-02:
When I need to connect nodes I have to hit this little cube on the node, which can be pretty hard, and then I need to drag the mouse and hit the input of another node exactly. This is pretty tedious.

The demo video shows snapping, but how can I do this.

Also, is it possible to make outputs and inputs bigger so they would be easier to click?

Another suggestion is, when I drag the mouse from node 1, I could hover over not only the cube representing the input on node 2, but also the name of the input or its value, so it would connect. Right now it connects only if i hover over the input cube.

astronouth7303, 2015-06-02:
Space bar causes it to snap.

+1 for more useful snapping

astronouth7303, 2015-06-02:
Moved to wishlist thread on the mailing list.

Highstaker, 2015-06-03:
Ah, thank you for the tip!

Heh, I've just discovered an instruction manual in docs XD

commit 4118b929 (2015-11-26): Merge branch 'awgrover-larger_ports' into develop

Closes #141
Closes #50
closed 2015-11-26 by mkeeter