mkeeter/antimony #51: Crashes on both of my Macs

fakepilot, 2015-06-02:
Getting a crash when trying to open the software. Version 8!

Process:               antimony [30674]
Path:                  /Applications/antimony.app/Contents/MacOS/antimony
Identifier:            com.impraxical.antimony
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           antimony [30674]
User ID:               509

Date/Time:             2015-06-02 13:40:37.756 +0200
OS Version:            Mac OS X 10.10.3 (14D136)
Report Version:        11
Anonymous UUID:        FBE6B5FD-ED07-2216-41B1-7D3F928D37B0

Sleep/Wake UUID:       E698B22F-700B-4B27-AE28-0F50416D39CA

Time Awake Since Boot: 230000 seconds
Time Since Wake:       18000 seconds

Crashed Thread:        0

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000002, 0x0000000000000000

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: /usr/local/Cellar/qt5/5.4.1/lib/QtWidgets.framework/Versions/5/QtWidgets
  Referenced from: /Applications/antimony.app/Contents/Frameworks/QtOpenGL.framework/Versions/5/QtOpenGL
  Reason: image not found

Binary Images:
    0x7fff6882a000 -     0x7fff68860837  dyld (353.2.1) <65DCCB06-339C-3E25-9702-600A28291D0E> /usr/lib/dyld
    0x7fff824cc000 -     0x7fff824cefff  libsystem_sandbox.dylib (358.20.5) <4CF77128-6BE0-3958-B646-707FA9CE61B2> /usr/lib/system/libsystem_sandbox.dylib
    0x7fff827ab000 -     0x7fff827c8fff  libsystem_kernel.dylib (2782.20.48) <EAFD7BD0-0C30-3E7D-9528-F9916BA0167C> /usr/lib/system/libsystem_kernel.dylib
    0x7fff82936000 -     0x7fff8293efff  libsystem_platform.dylib (63) <64E34079-D712-3D66-9CE2-418624A5C040> /usr/lib/system/libsystem_platform.dylib
    0x7fff83477000 -     0x7fff8347fffb  libcopyfile.dylib (118.1.2) <0C68D3A6-ACDD-3EF3-991A-CC82C32AB836> /usr/lib/system/libcopyfile.dylib
    0x7fff83f34000 -     0x7fff83f35ffb  libremovefile.dylib (35) <3485B5F4-6CE8-3C62-8DFD-8736ED6E8531> /usr/lib/system/libremovefile.dylib
    0x7fff83fa8000 -     0x7fff83fadfff  libsystem_stats.dylib (163.20.16) <FBC3F80F-A0FB-3BD6-9A7E-800DE45F092E> /usr/lib/system/libsystem_stats.dylib
    0x7fff83fae000 -     0x7fff83fe6fff  libsystem_network.dylib (412.20.3) <589A5F67-BE2A-3245-A181-0ECC9B53EB00> /usr/lib/system/libsystem_network.dylib
    0x7fff8407e000 -     0x7fff840f0fff  com.apple.framework.IOKit (2.0.2 - 1050.20.2) <09C0518C-90DF-3FC3-96D6-34D35F72C8EF> /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit
    0x7fff851f2000 -     0x7fff851f4fff  libquarantine.dylib (76.20.1) <7AF90041-2768-378A-925A-D83161863642> /usr/lib/system/libquarantine.dylib
    0x7fff85d2b000 -     0x7fff85d55ff7  libdispatch.dylib (442.1.4) <502CF32B-669B-3709-8862-08188225E4F0> /usr/lib/system/libdispatch.dylib
    0x7fff86938000 -     0x7fff8698cfff  libc++.1.dylib (120) <1B9530FD-989B-3174-BB1C-BDC159501710> /usr/lib/libc++.1.dylib
    0x7fff8699a000 -     0x7fff8699bfff  libsystem_secinit.dylib (18) <581DAD0F-6B63-3A48-B63B-917AF799ABAA> /usr/lib/system/libsystem_secinit.dylib
    0x7fff86ef7000 -     0x7fff86f6dfe7  libcorecrypto.dylib (233.1.2) <E1789801-3985-3949-B736-6B3378873301> /usr/lib/system/libcorecrypto.dylib
    0x7fff86f6e000 -     0x7fff86f6fff7  libsystem_blocks.dylib (65) <9615D10A-FCA7-3BE4-AA1A-1B195DACE1A1> /usr/lib/system/libsystem_blocks.dylib
    0x7fff8752f000 -     0x7fff87557fff  libxpc.dylib (559.20.9) <D35D0DB2-D7BD-3BE4-8378-062BFE545E1D> /usr/lib/system/libxpc.dylib
    0x7fff875a4000 -     0x7fff875a6fff  libsystem_configuration.dylib (699.1.5) <20F3B077-179D-3CB0-A3C1-C8602D53B4DB> /usr/lib/system/libsystem_configuration.dylib
    0x7fff87619000 -     0x7fff87619ff7  liblaunch.dylib (559.20.9) <FA89A113-696E-3271-8FE1-A0D7324E8481> /usr/lib/system/liblaunch.dylib
    0x7fff88a5a000 -     0x7fff88a5bfff  libDiagnosticMessagesClient.dylib (100) <2EE8E436-5CDC-34C5-9959-5BA218D507FB> /usr/lib/libDiagnosticMessagesClient.dylib
    0x7fff88ba4000 -     0x7fff88ba9ff7  libmacho.dylib (862) <126CA2ED-DE91-308F-8881-B9DAEC3C63B6> /usr/lib/system/libmacho.dylib
    0x7fff88d50000 -     0x7fff88d5bfff  libcommonCrypto.dylib (60061) <D381EBC6-69D8-31D3-8084-5A80A32CB748> /usr/lib/system/libcommonCrypto.dylib
    0x7fff8a0b9000 -     0x7fff8a145ff7  libsystem_c.dylib (1044.10.1) <86FBED7A-F2C8-3591-AD6F-486DD57E6B6A> /usr/lib/system/libsystem_c.dylib
    0x7fff8a146000 -     0x7fff8a14efff  libsystem_dnssd.dylib (561.1.1) <62B70ECA-E40D-3C63-896E-7F00EC386DDB> /usr/lib/system/libsystem_dnssd.dylib
    0x7fff8ae5e000 -     0x7fff8ae63ff7  libunwind.dylib (35.3) <BE7E51A0-B6EA-3A54-9CCA-9D88F683A6D6> /usr/lib/system/libunwind.dylib
    0x7fff8b4d4000 -     0x7fff8b4d6ff7  libsystem_coreservices.dylib (9) <41B7C578-5A53-31C8-A96F-C73E030B0938> /usr/lib/system/libsystem_coreservices.dylib
    0x7fff8b997000 -     0x7fff8b99dff7  libsystem_networkextension.dylib (167.1.10) <29AB225B-D7FB-30ED-9600-65D44B9A9442> /usr/lib/system/libsystem_networkextension.dylib
    0x7fff8be74000 -     0x7fff8be8aff7  libsystem_asl.dylib (267) <F153AC5B-0542-356E-88C8-20A62CA704E2> /usr/lib/system/libsystem_asl.dylib
    0x7fff8bea8000 -     0x7fff8beeeff7  libauto.dylib (186) <A260789B-D4D8-316A-9490-254767B8A5F1> /usr/lib/libauto.dylib
    0x7fff8beef000 -     0x7fff8befdff7  com.apple.opengl (11.1.2 - 11.1.2) <864B35BF-1E76-382B-8D5F-38C7282621E6> /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL
    0x7fff8c397000 -     0x7fff8c59146f  libobjc.A.dylib (647) <759E155D-BC42-3D4E-869B-6F57D477177C> /usr/lib/libobjc.A.dylib
    0x7fff8c961000 -     0x7fff8c991fff  libsystem_m.dylib (3086.1) <1E12AB45-6D96-36D0-A226-F24D9FB0D9D6> /usr/lib/system/libsystem_m.dylib
    0x7fff8c992000 -     0x7fff8c998fff  libsystem_trace.dylib (72.20.1) <840F5301-B55A-3078-90B9-FEFFD6CD741A> /usr/lib/system/libsystem_trace.dylib
    0x7fff8cb74000 -     0x7fff8cb90ff7  libsystem_malloc.dylib (53.1.1) <19BCC257-5717-3502-A71F-95D65AFA861B> /usr/lib/system/libsystem_malloc.dylib
    0x7fff8cbb9000 -     0x7fff8cbe4fff  libc++abi.dylib (125) <88A22A0F-87C6-3002-BFBA-AC0F2808B8B9> /usr/lib/libc++abi.dylib
    0x7fff8d05b000 -     0x7fff8d06cff7  libz.1.dylib (55) <88C7C7DE-04B8-316F-8B74-ACD9F3DE1AA1> /usr/lib/libz.1.dylib
    0x7fff8d167000 -     0x7fff8d178fff  libsystem_coretls.dylib (35.20.2) <6084A531-2523-39F8-B030-811FA1A32FB5> /usr/lib/system/libsystem_coretls.dylib
    0x7fff8d6e9000 -     0x7fff8d6e9ff7  libkeymgr.dylib (28) <77845842-DE70-3CC5-BD01-C3D14227CED5> /usr/lib/system/libkeymgr.dylib
    0x7fff8d74e000 -     0x7fff8d753fff  com.apple.DiskArbitration (2.6 - 2.6) <0DFF4D9B-2AC3-3B82-B5C5-30F4EFBD2DB9> /System/Library/Frameworks/DiskArbitration.framework/Versions/A/DiskArbitration
    0x7fff8d754000 -     0x7fff8d754ff7  libunc.dylib (29) <5676F7EA-C1DF-329F-B006-D2C3022B7D70> /usr/lib/system/libunc.dylib
    0x7fff8e597000 -     0x7fff8e5a0fff  libsystem_pthread.dylib (105.10.1) <3103AA7F-3BAE-3673-9649-47FFD7E15C97> /usr/lib/system/libsystem_pthread.dylib
    0x7fff9030e000 -     0x7fff904f3ff7  libicucore.A.dylib (531.48) <3CD34752-B1F9-31D2-865D-B5B0F0BE3111> /usr/lib/libicucore.A.dylib
    0x7fff90532000 -     0x7fff90539ff7  libcompiler_rt.dylib (35) <BF8FC133-EE10-3DA6-9B90-92039E28678F> /usr/lib/system/libcompiler_rt.dylib
    0x7fff90aae000 -     0x7fff90aafff3  libSystem.B.dylib (1213) <CCEC13A5-D0D9-31C5-B0B0-1C564B4A20A6> /usr/lib/libSystem.B.dylib
    0x7fff90ab0000 -     0x7fff90ad8fff  libsystem_info.dylib (459.20.1) <AEB3FE62-4763-3050-8352-D6F9AF961AE6> /usr/lib/system/libsystem_info.dylib
    0x7fff90aef000 -     0x7fff90e87ff7  com.apple.CoreFoundation (6.9 - 1153.18) <5C0892B8-9691-341F-9279-CA3A74D59AA0> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
    0x7fff90ec2000 -     0x7fff90ecbff7  libsystem_notify.dylib (133.1.1) <61147800-F320-3DAA-850C-BADF33855F29> /usr/lib/system/libsystem_notify.dylib
    0x7fff913d3000 -     0x7fff913d6ff7  libdyld.dylib (353.2.1) <9EACCA38-291D-38CC-811F-7E9D1451E2D3> /usr/lib/system/libdyld.dylib
    0x7fff918a3000 -     0x7fff918a7fff  libcache.dylib (69) <45E9A2E7-99C4-36B2-BEE3-0C4E11614AD1> /usr/lib/system/libcache.dylib

mkeeter, 2015-06-02:
Argh, it looks like `macdeployqt` didn't properly relink the libraries.  That's frustrating -- let me check it out.

mkeeter, 2015-06-02:
Okay, I updated the disk image (rebuilt with a newer version of `macdeployqt`).  Can you have a look and see if that fixed it?

fakepilot, 2015-06-02:
Nice! It works on my iMac now, cannot try the other one but should probably be fine. :-) Good speedy work!

closed 2015-06-02 by fakepilot