mkeeter/antimony #52: made a debian package.

revast, 2015-06-02:
ubuntu 14.04, qt54 libs bundled.
http://openartisthq.org/debian/trusty64/antimony_0.8.0-1~openArtist1_amd64.deb
enjoy

astronouth7303, 2015-06-02:
Do you have the source or instructions on how you built this package?

revast, 2015-06-03:
It's handmade.
see here: pastebin.com/f3xyxatf

joshuashort, 2015-06-03:
This is a good reason for me to try building on an Ubuntu VM.. I'll look into making a script at deploy/linux/mkdeb with what @revast posted.

joshuashort, 2015-06-07:
By all means, anyone else that can do this quicker probably should.. I'm not an expert on Debian (or generic Linux distro) packaging, but there are some good clues included in the instructions from @revast. I'm (slowly) working toward a packaging script that can be run from the (Linux) build directory..

To detect the needed libraries:

```
objdump -p antimony | grep NEEDED | tr -s ' ' | cut -d ' ' -f 3
```

These can each be piped to something like 

```
dpkg -S $libname | cut -d ':' -f 1 | head -n 1
```

to build a list of dependencies.. I noticed on an Ubuntu test system that some things, libpng12.so.0, for example, can yield multiple entries from dpkg:

```
libpng12-0:i386: /lib/i386-linux-gnu/libpng12.so.0.50.0
libpng12-0:i386: /lib/i386-linux-gnu/libpng12.so.0
libpng12-0:i386: /usr/lib/i386-linux-gnu/libpng12.so.0
```

So.. making a package for *every* Linux distro seems nearly impossible.. but maybe something can be found that generally works.. the package from @revast has some things in /opt and others in /usr..

Can anyone provide more insight/answers? For example, how can the DEBIAN/control file be easily generated?

joshuashort, 2015-06-07:
There's also the matter of which libraries get bundled and which are simply dependencies? It seems like any dependency of the form 'libQt5*' should be bundled.. is this right?

astronouth7303, 2015-06-07:
In my experience, it's generally easy enough to maintain the control file by hand. Finding documentation on how to, though, is somewhat trickier.

The other problem is that Debian and Ubuntu can sometimes have different package names. I don't think that's a problem for us yet.

revast, 2015-06-15:
Here is an updated version of the paste:  http://pastebin.com/UZErQaU5
I forgot to include the qt library plugins folder.
The debian package was also updated accordingly, and rebuilt with gcc4.8 instead of 4.9.

In general, to make most linux users happy, you could look at http://dispcalgui.hoech.net/, https://natron.inria.fr/ how they handle this.  dispcalgui uses opensuse build services, which once set up, generates packages for 

CentOS 6
CentOS 7
Debian 7 (Wheezy)
Debian 8 (Jessie)
Fedora 20
Fedora 21
Fedora 22
openSUSE 13.1
openSUSE 13.2
openSUSE Factory
openSUSE Tumbleweed
RedHat RHEL 6
RedHat RHEL 7
Ubuntu 12.04 (Precise)
Ubuntu 14.04 (Trusty)
Ubuntu 14.10 (Utopic)
Ubuntu 15.04 (Vivid) 

Which has the advantage that this then is also a package repository for every of the mentioned distribution, 32 and 64bit.

Plus they use Zero Install... just take a look at it. It even works on osx.

The natron guys use a different approach, they provide a selfcontained .tgz download which contains an installer a la windows, 64bit only.

Then there is http://portablelinuxapps.org/, which uses fuse to pack all dependencies into one package, a very elegant solution, like mac .app packages.. 32 or 64bit.

Even a commercial software for doing this is available, called Ermine. http://www.magicermine.com/. "An executable prepared on an x86 can run on x86_64 without 32-bit compatibility libraries."

As well I want to mention its also possible to use docker, as this also works fine for packasgin single GUI applications.

astronouth7303, 2015-07-05:
Debian Testing and Ubuntu Vivid both have QT5 now, so building this package should be significantly easier.

gcb, 2015-09-25:
anyone working on it? i may like to offer to be a maintainer of deb (and hopefully others, but... baby steps) package.

mkeeter, 2015-09-26:
I don't think anyone is actively working on this at the moment -- volunteering as a deb maintainer would definitely be welcome.  Now that Qt 5.4 is available without jumping through hoops (see #97), this should be easier to do.

tiagovaz, 2018-08-27:
Hi, I´ve filled an ITP and I´m working to get an official Debian package of antinomy in a few days (hours?).

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=907435

revast, 2018-08-28:
nice!

tiagovaz, 2018-08-31:
The package has been uploaded 3 days ago but is still waiting for ftp-master approval: https://ftp-master.debian.org/new/antimony_0.9.3-1.html

For the record: antimony packaged had to depend on python3-all-dev and libqt5opengl5-dev to build in a clean Sid chroot, which are not currently listed in the upstream control file.

tiagovaz, 2018-11-15:
The package has finally been accepted and is officially in Debian now. I think you can close this bug.

mkeeter, 2018-11-15:
Cool!

closed 2018-11-15 by mkeeter

revast, 2018-11-15:
nice!! official packages are the way to go!