mkeeter/antimony #53: Program fails on an assert

Highstaker, 2015-06-03:
Maybe it's not a big deal, since it's just an assert fail, but 

https://github.com/mkeeter/antimony/blob/develop/src/render/render_task.cpp#L67

this assert seems to fail when I try adding a 2D node (circle, text, etc.). But it is okay if I try 3D one.

Just to let you know

astronouth7303, 2015-06-04:
Can you give more detail how to reproduce this?

mkeeter, 2015-06-05:
I can reproduce this by manually creating a shape with semi-infinite bounds, e.g.
```
import fab
title('Assert')
output('s', fab.types.Shape('X', -1, -1, -float("inf"), 1, 1, 0))
```
This is a bit contrived, but I'll rethink the assertion (we probably just default to 2D if either Z bound is infinite).

mkeeter, 2015-06-05:
Okay, fixed in 7466691f91acd7d02a47480ec61f9ee37c9a3703
(let me know if further issues arise).

closed 2015-06-05 by mkeeter