mkeeter/antimony #54: Handling Enter on input and layout messing

Highstaker, 2015-06-03:
As a Blender user, I often tend to press Enter after I input the data into the node, which is not needed in this program. I'm fine with that. But, when I press Enter, the cursor moves to a new line. Moreover, when I'm trying to Backspace to remove these newlines, the cursor seems to jump randomly to the start or... I don't know where. Additionally, it screws the layout of the node up, rearranging the data in the node weirdly. Like, I was inputting X, and suddenly it appears at the bottom of the node.

I'm not sure if multiline is needed in node input fields, maybe it should be made into single line instead, and Enter should be made to do nothing?

mkeeter, 2015-06-03:
`Enter` should probably just give up focus (signalling that you're done editing).  If you'd like to patch this in yourself, check out the event filter [here](https://github.com/mkeeter/antimony/tree/develop/src/ui/canvas/inspector/inspector_text.cpp#L147-L165).

mgold, 2015-06-03:
You'd be amazed how infrequently some OSS maintainers will tell new contributors where to look.

joshuashort, 2015-06-03:
The line-block links are incredibly helpful! It's nice to get so much help on how to make changes.

I think I broke some stuff with my git client.. but I made a pull request https://github.com/mkeeter/antimony/pull/45 that was only supposed to cover the Return key behavior when editing text on canvas items. Really, it's just an or statement tacked onto the tab key filter..

joshuashort, 2015-06-03:
I'm amazed at how nice and convenient the hints are :+1:

This simple pull request changes the 'return' key behavior to act like tab: https://github.com/mkeeter/antimony/pull/55

mkeeter, 2015-06-03:
Okay, closed with #55

closed 2015-06-03 by mkeeter

Highstaker, 2015-06-04:
Yeah, I'm really impressed how quickly it got resolved!