mkeeter/antimony #57: Iterate (now Array) with transform function?

dylanmc, 2015-06-04:
I'd like to make a spiral staircase by iterating a step around a cylinder. I can use the Array (polar) to wrap around the cylinder at one altitude, but I'd really like to apply a transformation on each array step. Having done this, I'd want to be able to do something more than once around the polar axis. (say 3 twists...)

Am I missing an elegant way to make a spiral staircase as described, or is something new like Array-with-function needed to do it?

mkeeter, 2015-06-04:
This is a pretty easy job for a custom script, something like:

```py
import fab

import functools
import operator
from fab.shapes import rotate, move

title('Array (staircase)')

input('shape', fab.types.Shape)
input('x', float)
input('y', float)
input('dz', float)
input('n', int)
input('N', int)

p = functools.reduce(operator.or_,
            [move(rotate(shape, 360./N * i, x, y),
                  0, 0, dz * i)
             for i in range(n)])

output('array', p)

# UI
fab.ui.wireframe([
    (math.cos(i/36. * 2*math.pi) * 0.3 + x,
     math.sin(i/36. * 2*math.pi) * 0.3 + y, 0)
    for i in range(36)], close=True, color=fab.color.teal)

fab.ui.point(x, y, 0, color=fab.color.teal)
```

dylanmc, 2015-06-04:
Great example! I couldn't have created that on my own, but with examples like this, wow -- I'm more and more amazed by Antimony. Closed.

closed 2015-06-04 by dylanmc

mbd-shift, 2020-09-18:
http://www.mattkeeter.com/projects/antimony/3/

**Please** add on the website that iterate is now array, I've been searching for it forever