mkeeter/antimony #58: Tab on the last entry in node

Highstaker, 2015-06-04:
When I press Tab while the bottom focusable entry is in focus - it simply deselects. If I press Tab again, it focuses on the bottommost entry again. I was expecting it to jump to the first one.

I thought about playing around with this function https://github.com/mkeeter/antimony/blob/develop/src/ui/canvas/inspector/inspector.cpp#L249

but it left me confused because:

a) I don't know how to get the first row
b) Most nodes have a "Shape" entry as the bottommost, and it is not focusable, yet this: https://github.com/mkeeter/antimony/blob/develop/src/ui/canvas/inspector/inspector.cpp#L264
returns true and it tries to focus on it nevertheless. And I can't find a way to make it  jump over unfocusable entries.

mgold, 2015-06-04:
Maybe I'm naive here, but the simplest possible solution seems to be keeping an array of entries and an index that is incremented/decremented and then `%=` by the length of the array.

Highstaker, 2015-06-04:
I think I know how to solve the issue a) I had. I can just save the first Datum into the variable and use it to access the first InspectorRow when I need it. 

Now I need to make it filter the unfocusable fields. And then - add detection whether the last field is selected or not and make it jump using the firstrow I saved.

PS. I'm just writing this here so I wouldn't forget my line of thought.

Highstaker, 2015-06-04:
https://github.com/Highstaker/antimony/commit/dc66a977bc5b260011c0a64e16cf61e28eba499f

Started doing this. With this commit it seems to work when I have a node that has no Shape entry, in other words, all entries are focusable. Still need to figure out how to detect unfocusable rows.

joshuashort, 2015-06-05:
For what it's worth, the same general thing happens when shift-tabbing past the first item.

Highstaker, 2015-06-05:
@joshuashort Yeah, pretty much. ```NodeInspector::focusPrev``` gotta be modified as well.

https://github.com/Highstaker/antimony/commit/bae3ecff1b8a4f9edc77e7e1be48e53dcf6f6c08

Added a new commit, which seems to fix that for Tab, I just detect whether the row has an input or not (by calling the associated Datum's ```hasInput()```). But now it stumbles if there is a row to which an input is connected, because it becomes unfocusable. This is one thing to consider.

Another thing, something is not right with, say, Extrude node or Reflect(XZ), it doesn't work with these. It seems that it stumbles upon invisible (or private?) inputs marked as ```input('_x', type)``` in their scripts. Need to find a way to detect that as well.

Highstaker, 2015-06-05:
> But now it stumbles if there is a row to which an input is connected, because it becomes unfocusable. This is one thing to consider.

This is probably solved by this commit

https://github.com/Highstaker/antimony/commit/a805d63c244bd2d8d54df6b7a5ebd1d47907f33e

Now, the second problem - hidden rows.

Highstaker, 2015-06-05:
On the other hand, ```show_hidden``` seems to be not needed. ```isDatumHidden(d)``` manages everything as needed. Probably...

https://github.com/Highstaker/antimony/commit/8d152b78ba674f060efe6b5cce5d5a598501bb06

Now Tabbing seems to work for all nodes. Or not? I haven't noticed any stumbles yet.

Highstaker, 2015-06-05:
Modified ```focusPrev``` here 

https://github.com/Highstaker/antimony/commit/4e90569decc3fff1be5044fe467471a730015da6

I wonder if it is okay to pull now.

mkeeter, 2015-06-07:
Okay, I've ended up doing a slightly cleaner implementation in e20af58a8efa5a0e5c965dab4fb9f2b20ff81434 -- let me know if you see any issues.

closed 2015-06-07 by mkeeter