mkeeter/antimony #60: Nodes with unicode characters may not function properly

C-o-r-E, 2015-06-04:
Specifically the [2D → 3D] node does not function if UTF-8 support is lacking. Antimony will start normally and display the proper name in the add menu however upon mousing over [2D → 3D], no sub-menu appears.

A workaround on linux can be to specify the locale with the LANG environment variable:

```
$ LANG=en_US.UTF8 ./antimony 
```

Perhaps it would be a good idea to detect this kind of problem and warn the user.

astronouth7303, 2015-06-04:
Hm. Unfortunately, that could interfere with unicode on the rest of filesystem.

I can see 3 ways of dealing with this:
1. Use different encodings when reading from the filesystem for different reasons (complicating code)
2. Make sure it gets transcoded on installation (complicating installation)
3. Tell users that it's 2015 and use UTF-8 (complicating support)

astronouth7303, 2015-06-04:
On what systems is this reproducible, and how do you do it?

C-o-r-E, 2015-06-04:
I think any modern Linux system with LOCALE set to "C" will have the same result. Or something that doesn't support the arrow character in this example.

```
$ env | grep LANG
LANG=C
```

astronouth7303, 2015-06-04:
`$ LANG=C ./antimony` does not reproduce the problem for me on develop.

mkeeter, 2015-06-04:
It reproduces for me on Linux, not on Mac.

mkeeter, 2015-06-04:
Should be fixed now; re-open if you're still having trouble.

closed 2015-06-04 by mkeeter

sagitter, 2015-11-23:
Hi,

this problem is still present in Fedora 23; to me, it happens for ```2d->3D``` and ```3D->2D``` menus:

```
$ env | grep LANG
LANG=en_US.UTF-8
GDM_LANG=en_US.UTF-8
$ LANG=en_US.UTF-8 antimony
 ```

dkadish, 2015-12-08:
Second sagitter on Fedora 23. But for me it's all of the context (Add) menus. Nothing shows up under 2D, 3D, 2D > 3D, 3D > 2D, or CSG.

```
$ env | grep LANG
LANG=en_CA.UTF-8
GDM_LANG=en_CA.UTF-8
```

I tried running it with a bunch of different LANG env variables with no success.

mcarlo95, 2016-04-11:
Same problem here on Fedora 23.