mkeeter/antimony #61: Line Node Request

bbmech7, 2015-06-04:
Perhaps it come from my extensive use of Solidworks but i would like to see a 2 point line node and a mid point line node. 

Also some manner of selecting or linking to the midpoint of a give shape or line would be awesome.

One more thing... would it be possible to convert a node after it is placed. for instance say you've created a circle and placed it where you want but realize you need center and radius rather than the other one. not a huge deal but the ability to switch might help with workflow.

Also a question, is there plans of  incorporating the Cam portion of  kokopelli into antimony? cause that would be awesome. specially if it could be setup to handle 3d printer Gcode this could almost be a one stop deal model to sd to print. 

Last thing i promise, I briefly read through sections of the thesis paper that contains info on node creation. Is there plans to have a more detailed write up about how the syntax of those line are handled.  i saw the table with the operator and variable definitions but i was a bit confused with how the string that was shown related to the equation to the side.

astronouth7303, 2015-06-04:
1. What would you want to do with a line? Implementing lines would be very difficult for Antimony (See: Functional representation and Implied surfaces)

2. Node conversion isn't currently implemented at all, or even on roadmap. A lot of node infrastructure would have to be added to implement it reliably.

3. It hasn't been discussed so far. AFAIK, there's plenty to do on the CAD side, and there's lots of things that can slice an STL.

4. "sparse prefix notation" is a bit of a bear. I hear someone is working on it.

bbmech7, 2015-06-04:
Aline node would be useful when drawing odd or complex geometries. i understand full well that the same shapes could be made with a variety of simple shapes and a combination of Boolean operations but i very much enjoy working with clean concise drawings that are free of clutter.  

For instance if i were to be designing a spindle that needs a minimum of 2 bearing surfaces, a shoulder for location, a thread and a diameter for a pulley or drive mechanism. I would have a fairly cluttered design area with many "phantom lines" leftover after my revolve feature is done.  In this instance with a line node i could add a series of lines connected end to end to produce the profile needed then revolve the resulting shape. 

Again it has a lot to do for with what i am accustomed to and it would be a nice feature.  

That all being said, is there a way turn off the visibility of the lines from the primitive shapes once you are satisfied with the solid body you have? I have a terrible habit of building entire machines in one multibody part file and breaking them out after I am satisfied with the design.

astronouth7303, 2015-06-04:
Antimony has no concept of paths at this time (see the discussion of #31). 

Being able to hide UI of primitives would really help clean up the display, and subparts is an important part of the CAD workflow. Forwarded to the wishlist.

closed 2015-06-04 by astronouth7303

joshuashort, 2015-06-05:
Maybe I was doing things 'wrong', but I was playing around with antimony and came across some things that it seemed like a line node would be great for.. and this may be going too far down the 'wrong' line of thought, but maybe it's useful to be able to explicitly make something as 'reference' vs. 'printable'.. maybe some nodes only exist to make the creation of the final product easier?

astronouth7303, 2015-06-05:
Only shapes connected to an Export node are actually exported.

If you're making custom nodes, you can add lines and points as UI. (See `fab.ui`.)

joshuashort, 2015-06-05:
I figured there was a better way -- thanks for the pointer!

astronouth7303, 2015-06-05:
Yeah, one of the things I've slowly been working on is documenting the Python half of things.

bbmech7, 2015-06-05:
yes i will definitely play with the concept of line nodes even if just for the sake of reference lines.

mkeeter, 2015-06-05:
You can make a reference line with a custom script, something like

```py
import fab

title('Line')

input('x0', float)
input('y0', float)
input('x1', float)
input('y1', float)

fab.ui.wireframe([(x0, y0, 0), (x1, y1, 0)])
fab.ui.point(x0, y0, 0)
fab.ui.point(x1, y1, 0)
```

(though I'm not sure what to do with it afterwards)