mkeeter/antimony #65: Not asked to Save when closing the last window

astronouth7303, 2015-06-05:
If you manually just 'x' out of the windows, Antimony exits when the last one is closed without prompting to save a changed document.

Highstaker, 2015-06-05:
+1. Quitting with Ctrl+Q does ask, but not all people quit that way.

Also, it doesn't ask it when I do it via "RMB -> Close" from a taskbar in Ubuntu (because it groups windows of the same app). It is essentially the same problem though.

Highstaker, 2015-06-05:
I think this could be handled this way:
https://github.com/Highstaker/antimony/commit/d8255e7339c5897a4e372a9fa196bfd884eb4117
After a cleanup:
https://github.com/Highstaker/antimony/commit/3a4e06d003800cf30fcfac58c6189eb352f55fc8

Had to add a new static variable to ```MainWindow``` that counts the amount of opened windows. Thought about using ```App::topLevelWindows()```, but it seems to update slowly, so it's not reliable.

Highstaker, 2015-06-07:
https://github.com/Highstaker/antimony/commit/b576ecee98c3d96e352176f95ea70ca25d9abb86

Made another message box, which actually prompts whether you want to save or not.
Need a way to see whether user pressed Cancel **in SaveAs window**, because if they *saved* - the program should quit after saving, and if they *pressed Cancel* - it should not quit. Right now it doesn't quit in both cases.

Highstaker, 2015-06-07:
https://github.com/Highstaker/antimony/commit/dcc2a6e269f2b49215622d41685c36f5ea669fca#diff-2508dc1897fe294f5e3c9cfa438357caL154

All done here. Seems to work, not entirely sure though.

ApostolosB, 2016-01-10:
Is this ready to be included in the next release? Could you make a pull request?

ApostolosB, 2016-01-27:
Paging @mkeeter cause it is an annoying(ish) bug.

mkeeter, 2016-01-28:
Yep.  I'd like @Highstaker to open up a PR for the changes (rather than just bringing them in unilaterally).  If they don't do so, I'll make sure to fix this in the never-ending UI rewrite.

Highstaker, 2016-01-29:
I can try to open a PR, but I've done it a loooong while ago, so it might not work with the current version.

Okay, here goes nothing! https://github.com/mkeeter/antimony/pull/146

mkeeter, 2016-01-30:
Fixed in 860d60231391418dfa84908dc845bcd2cf540f0f

closed 2016-01-30 by mkeeter