mkeeter/antimony #70: Building using mac ports

thesamprice, 2015-06-05:
Was trying to build using mac ports.
I added the following to the antimony.pro file.
"""
macx {
 contains(QMAKE_CXX, /usr/bin/clang++) {
 message(Using LLVM libc++)
 QMAKE_CXXFLAGS += -stdlib=libc++
 QMAKE_LFLAGS += -lc++
 QMAKE_CXXFLAGS += $$system(python3.4-config --include)
 }
}
"""
to get it somewhat compiling.
Started running into the following errors after that.
"""
../src/graph/datum/datum.cpp:86:26: error: 'destroyed' is a protected member of 'QObject'
    connect(link, &Link::destroyed, this, &Datum::connectionChanged);
                         ^
/opt/local/include/QtCore/qobject.h:270:10: note: declared protected here
    void destroyed(QObject * = 0);
         ^
../src/graph/datum/datum.cpp:86:5: error: no matching member function for call to 'connect'
    connect(link, &Link::destroyed, this, &Datum::connectionChanged);
    ^~~~~~~
"""

mkeeter, 2015-06-05:
This may be a Qt version issue -- are you building with Qt 5?

thesamprice, 2015-06-05:
I just realized i was building with 4.8 and not qt 5.  Thats probably the issue.  I have a lot of stuff dependent on 4.8 (mainly gnuradio), so I will prob hold off for a while.  Feel free to close the issue.

closed 2015-06-05 by mkeeter