mkeeter/antimony #71: Native Windows build

mkeeter, 2015-06-05:
I've started working on a native Windows build in [this branch](https://github.com/mkeeter/antimony/tree/feature/win32-build)
(instructions emerging [here](https://github.com/mkeeter/antimony/blob/feature/win32-build/BUILDING.md#windows)).

Current blockers:
-  `libpng` is being built without any symbols,
- We're missing a few important symbols from `python3.dll`
Both of these are causing the linker to fail.

@s7726, have you gotten any farther with a native build?

s7726, 2015-06-08:
I have not been able to get to this no.

carlosjln, 2015-06-23:
I wish this build comes soon, i tried building one on xubuntu and it failed (i'm a windows user)

Betosaurus, 2015-07-28:
I am interested in a Windows build too.

DeadlyDad, 2015-10-13:
Any news on a Windows binary?  I'm using the v0.8.0 on a Xubuntu virtual machine, but that's a little sketchy.

DeadlyDad, 2015-10-14:
I am absolutely flat broke, but if someone [either has or wants to make some money.....](https://www.bountysource.com/issues/20680913-native-windows-build)

mkeeter, 2015-10-14:
No news on Windows binaries.  I've had a few pings from people interested, but no solid followup yet.

Any reason why you're still on 0.8.0?  That's a pretty old release...

DeadlyDad, 2015-10-14:
1. ...which is why the bounty link.  Those who want it and have a few bucks to spare provide the financial incentive for someone to take the time to make it happen.
2. I followed the compiling instructions that people put up in a VM Xubuntu environment, but could never get it to work, and, as I haven't done any real coding in *years*, couldn't figure out why it wouldn't.  The only reason that I have been able to use Antimony is that someone posted a 0.8.0 .deb file.  (Even that needed the compiling prerequisites to actually install it.)  If someone would put together a .deb of the current release, that would be *GREAT*.  BTW, is it possible to export a Collada or .obj file yet?

TLC123, 2016-03-03:
I seeded an tiny bounty on Bountysource. Anyone else up for a buck or two?

DeadlyDad, 2017-01-02:
Anybody willing to have mercy on us poor Windows shlubs?  Some of just don't have the time to learn a complete coding environment to compile a single program.

mkeeter, 2017-01-18:
I had my annual fight against Windows; unlike in past years, I came out victorious:

[image]

Everything is merged into the `develop` branch, with instructions [here](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md#windows-experimental).  Building is a fairly hands-on process, but those instructions should be clear.

Building an installer is a whole different can of worms, and something I'll punt on for now.

@DeadlyDad – want to try following those instructions and see if you can get something running?

carlosjln, 2017-01-18:
Awesome! I'll give it a try sometime soon :)

DeadlyDad, 2017-01-22:
Thanks!  Trying...

Issues:

   1. Instructions say '/mingw64/home/$USERNAME'.  Should be
   '\msys64\home\$USERNAME'?
   2. Couldn't compile Boost.  Help!  Error:

spamt@DESKTOP-D1M1OJ0 MSYS ~/boost_1_63_0
>
> $ cd ~/boost_1_63_0
>
>
>> spamt@DESKTOP-D1M1OJ0 MSYS ~/boost_1_63_0
>
> $ ./bootstrap.sh
>
> Building Boost.Build engine with toolset ...
>
> Failed to build Boost.Build build engine
>
> Consult 'bootstrap.log' for more details
>
>
>> spamt@DESKTOP-D1M1OJ0 MSYS ~/boost_1_63_0
>
> $ ls
>
> boost      boost.png        boostcpp.jam   bootstrap.log  doc
>>  index.html  Jamroot  LICENSE_1_0.txt  rst.css  tools
>
> boost.css  boost-build.jam  bootstrap.bat  bootstrap.sh   index.htm
>>  INSTALL     libs     more             status
>
>
>> spamt@DESKTOP-D1M1OJ0 MSYS ~/boost_1_63_0
>
> $ cat bootstrap.log
>
> ###
>
> ### Using 'kcc' toolset.
>
> ###
>
> rm -rf bootstrap
>
> mkdir bootstrap
>
> KCC -o bootstrap/jam0 command.c compile.c constants.c debug.c execcmd.c
>> frames.c function.c glob.c hash.c hdrmacro.c headers.c jam.c jambase.c
>> jamgram.c lists.c make.c make1.c object.c option.c output.c parse.c
>> pathsys.c regexp.c rules.c scan.c search.c subst.c timestamp.c variable.c
>> modules.c strings.c filesys.c builtins.c class.c cwd.c native.c md5.c
>> w32_getreg.c modules/set.c modules/path.c modules/regex.c
>> modules/property-set.c modules/sequence.c modules/order.c execunix.c
>> fileunix.c pathunix.c
>
> Usage: : KCC [--version] [--help] command ..
>
>
>>

mkeeter, 2017-01-22:
>1. Instructions say '/mingw64/home/$USERNAME'.  Should be '\msys64\home\$USERNAME'?

Okay, clarified in 8fd10a9102f4d43afaab4add94dd183dcc9672ca

>2. Couldn't compile Boost.  Help!

I'm not totally sure what's going on here, but it looks like you're in the wrong MSYS shell.  You should be opening the "MSYS2 MinGW 64-bit" shell application, which has a prompt containing `MINGW64`:
- My prompt is `mkeeter@MATTHEWKEETF381 MINGW64 ~/boost_1_63_0`
- Yours is `spamt@DESKTOP-D1M1OJ0 MSYS ~/boost_1_63_0`

Can you try opening the correct shell and walking through the directions again?

DeadlyDad, 2017-01-23:
>    1. Couldn't compile Boost. Help!

1. Thanks.
   2. :facepalm:  I didn't pay enough attention.  It just popped up a
   console, and I assumed that it was the right one.  Thanks for setting me
   right.
   3. Got much further, but there was a *HUGE* wall of text that looked
   like errors(?) as shown in the first quote, but seemed to produce...
   something.

>    1. ...patience...
>    2. ...patience...
>    3. ...found 1861 targets...
>    4. ...updating 78 targets...
>    5. common.mkdir stage
>    6. common.mkdir stage\lib
>    7. common.mkdir bin.v2\libs
>    8. common.mkdir bin.v2\libs\python
>    9. common.mkdir bin.v2\libs\python\build
>    10. common.mkdir bin.v2\libs\python\build\gcc-mingw-6.3.0
>    11. common.mkdir bin.v2\libs\python\build\gcc-mingw-6.3.0\debug
>    12. common.mkdir
>    bin.v2\libs\python\build\gcc-mingw-6.3.0\debug\link-static
>    13. common.mkdir
>    bin.v2\libs\python\build\gcc-mingw-6.3.0\debug\link-static\threading-multi
>    14. gcc.compile.c++
>    bin.v2\libs\python\build\gcc-mingw-6.3.0\debug\link-static\threading-multi\numeric.o
>    15. In file included from ./boost/python/detail/is_xxx.hpp:8:0,
>    16.                  from ./boost/python/detail/is_auto_ptr.hpp:9,
>    17.                  from
>    ./boost/python/detail/copy_ctor_mutates_rhs.hpp:8,
>    18.                  from ./boost/python/detail/value_arg.hpp:7,
>    19.                  from ./boost/python/object/forward.hpp:13,
>    20.                  from ./boost/python/object/pointer_holder.hpp:16,
>    21.                  from ./boost/python/to_python_indirect.hpp:10,
>    22.                  from
>    ./boost/python/converter/arg_to_python.hpp:10,
>    23.                  from ./boost/python/call.hpp:15,
>    24.                  from ./boost/python/object_core.hpp:14,
>    25.                  from ./boost/python/object.hpp:9,
>    26.                  from ./boost/python/tuple.hpp:10,
>    27.                  from ./boost/python/numeric.hpp:10,
>    28.                  from libs\python\src\numeric.cpp:6:
>    29. ./boost/python/detail/is_auto_ptr.hpp:17:40: warning:
>    'template<class> class std::auto_ptr' is deprecated
>    [-Wdeprecated-declarations]
>    30.  BOOST_PYTHON_IS_XXX_DEF(auto_ptr, std::auto_ptr, 1)
>    31.                                         ^
>    32. ./boost/detail/is_xxx.hpp:20:4: note: in definition of macro
>    'BOOST_DETAIL_IS_XXX_DEF'
>    33.     qualified_name< BOOST_PP_ENUM_PARAMS_Z(1, nargs, T) >        \
>    34.     ^~~~~~~~~~~~~~
>    35. ./boost/python/detail/is_auto_ptr.hpp:17:1: note: in expansion of
>    macro 'BOOST_PYTHON_IS_XXX_DEF'
>    36.  BOOST_PYTHON_IS_XXX_DEF(auto_ptr, std::auto_ptr, 1)
>    37.  ^~~~~~~~~~~~~~~~~~~~~~~
>    38. In file included from
>    C:/msys64/mingw64/include/c++/6.3.0/memory:81:0,
>    39.                  from ./boost/config/no_tr1/memory.hpp:21,
>    40.                  from ./boost/get_pointer.hpp:14,
>    41.                  from ./boost/python/object/pointer_holder.hpp:11,
>    42.                  from ./boost/python/to_python_indirect.hpp:10,
>    43.                  from
>    ./boost/python/converter/arg_to_python.hpp:10,
>    44.                  from ./boost/python/call.hpp:15,
>    45.                  from ./boost/python/object_core.hpp:14,
>    46.                  from ./boost/python/object.hpp:9,
>    47.                  from ./boost/python/tuple.hpp:10,
>    48.                  from ./boost/python/numeric.hpp:10,
>    49.                  from libs\python\src\numeric.cpp:6:
>    50. C:/msys64/mingw64/include/c++/6.3.0/bits/unique_ptr.h:49:28: note:
>    declared here
[... 91 lines cut ...]
>    37. spamt@DESKTOP-D1M1OJ0 MINGW64 ~
>    38. $ cd antimony
>    39.
>    40. spamt@DESKTOP-D1M1OJ0 MINGW64 ~/antimony
>    41. $ mkdir build
>    42.
>    43. spamt@DESKTOP-D1M1OJ0 MINGW64 ~/antimony
>    44. $ cd build
>    45.
>    46. spamt@DESKTOP-D1M1OJ0 MINGW64 ~/antimony/build
>    47. S Makefiles" ..boost_1_63_0/
>    BOOST_LIBRARYDIR=~/boost_1_63_0/stage/ cmake -G"MSY
>    48. CMake Error: CMake was unable to find a build program
>    corresponding to "MSYS Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You
>    probably need to select a different build tool.
>    49. CMake Error: CMake was unable to find a build program
>    corresponding to "MSYS Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You
>    probably need to select a different build tool.
>    50. CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
>    51. CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
>    52. CMake Error: CMAKE_AR was not found, please set to archive program.
>    53. -- Configuring incomplete, errors occurred!
>    54.
>
>

mkeeter, 2017-01-24:
Okay, I think the issue is the path to `cmake`.  Can you try invoking it as 
```BOOST_ROOT=~/boost_1_63_0/ BOOST_LIBRARYDIR=~/boost_1_63_0/stage/ /mingw64/bin/cmake -G"MSYS Makefiles" ..```
instead?

Updated the docs in commit 48fdfc608420f2d67779cd2d04f6a9227ec1c518

DeadlyDad, 2017-01-25:
spamt@DESKTOP-D1M1OJ0 MINGW64 ~/antimony/build
$ BOOST_ROOT=~/boost_1_63_0/ BOOST_LIBRARYDIR=~/boost_1_63_0/stage/ cmake
-G"MSYS Makefiles" ..
CMake Error: CMake was unable to find a build program corresponding to
"MSYS Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to
select a different build tool.
CMake Error: CMake was unable to find a build program corresponding to
"MSYS Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to
select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_AR was not found, please set to archive program.
-- Configuring incomplete, errors occurred!

On 24 January 2017 at 15:44, Matt Keeter <notifications@github.com> wrote:

> BOOST_ROOT=~/boost_1_63_0/ BOOST_LIBRARYDIR=~/boost_1_63_0/stage/
> /mingw64/bin/cmake -G"MSYS Makefiles" .

DeadlyDad, 2017-01-25:
(Sorry.  Pasted the wrong one.)

spamt@DESKTOP-D1M1OJ0 MINGW64 ~/antimony/build
$ BOOST_ROOT=~/boost_1_63_0/ BOOST_LIBRARYDIR=~/boost_1_63_0/stage/
/mingw64/bin/cmake -G"MSYS Makefiles" ..
CMake Error: CMake was unable to find a build program corresponding to
"MSYS Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to
select a different build tool.
CMake Error: CMake was unable to find a build program corresponding to
"MSYS Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to
select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_AR was not found, please set to archive program.
-- Configuring incomplete, errors occurred!

On 25 January 2017 at 07:05, Keith O <spamtrap007@gmail.com> wrote:

> spamt@DESKTOP-D1M1OJ0 MINGW64 ~/antimony/build
> $ BOOST_ROOT=~/boost_1_63_0/ BOOST_LIBRARYDIR=~/boost_1_63_0/stage/ cmake
> -G"MSYS Makefiles" ..
> CMake Error: CMake was unable to find a build program [...]

mkeeter, 2017-01-31:
Okay, how about running `pacman -Q` in a `MSYS MinGW64` shell to list installed packages?  

When I do so, I get [this list](https://gist.github.com/mkeeter/6f4b540047180670909ef9930e69edca).  Can you `diff` that with your results and see what's missing?

DeadlyDad, 2017-02-03:
Hmmm... 'Twould appear I am missing some very important packages.
https://www.diffchecker.com/S6gfEbz1

mkeeter, 2017-02-04:
I think the important one that you're missing is `make` – can you install it (`pacman -S make`) and see if `cmake` becomes happy?

eltercerlugar, 2017-02-05:
Hi all. Im a Windows 10 user, with a enough clean system installation. I follow the [instructions to build the Windows enviroment](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md) to work with Antimony, and after follow all the steps i could not compile the Antimony package.

The problem seemed to be the Boost package, so i found a package that, added to the list of packaged installed with pacman, make my antimony runs in windows.

I added this command:

Pacman -S base-devel

before the "clone and build antimony" part, and i could finally avoid the error after prompt "BOOST_ROOT=~/boost_1_63_0/ BOOST_LIBRARYDIR=~/boost_1_63_0/stage/ /mingw64/bin/cmake -G"MSYS Makefiles" .."

Hope this helps anyone.

seantapscott, 2017-03-07:
I got the same cmake configuration errors as DeadlyDad on the cmake step, and had to run 
`pacman -s make` as well.   Not entirely successful yet, but still working on it.  Really excited, thank  you mkeeter.

seantapscott, 2017-03-07:
And its working.  Not entirely sure what to do next, but boom!

roboter, 2018-09-04:
@seantapscott could you write instruction how to build, on clean PC, 
with all versions. 
I have tried many times and failed.