mkeeter/antimony #75: Building doesn't work out of the box for all Linuxes

jayay, 2015-06-06:
Hello guys!
I tried to build antimony on Gentoo and I had similar problems as in #18.
I changed a few lines until everything built fine.
This is what I changed: 

```diff
diff --git a/qt/fab.pri b/qt/fab.pri
index a54813b..3653a8d 100644
--- a/qt/fab.pri
+++ b/qt/fab.pri
@@ -54,7 +54,7 @@ DEFINES += '_STATIC_= '
 
 linux {
     QMAKE_CFLAGS += -std=gnu99
-    QMAKE_CXXFLAGS += $$system(/usr/bin/python3-config --includes)
+    QMAKE_CXXFLAGS += $$system(/usr/bin/python3.4-config --includes)
     LIBS += -lpng
 }
 
diff --git a/qt/shared.pri b/qt/shared.pri
index 206d0d7..88fecba 100644
--- a/qt/shared.pri
+++ b/qt/shared.pri
@@ -49,17 +49,17 @@ macx {
 }
 
 linux {
-    QMAKE_CXXFLAGS += $$system(/usr/bin/python3-config --includes)
-    QMAKE_LFLAGS   += $$system(/usr/bin/python3-config --ldflags)
+    QMAKE_CXXFLAGS += $$system(/usr/bin/python3.4-config --includes)
+    QMAKE_LFLAGS   += $$system(/usr/bin/python3.4-config --ldflags)
 
     # Even though this is in QMAKE_LFLAGS, the linker is picky about
     # library ordering (so it needs to be here too).
-    LIBS += -lpython3.4m
+    LIBS += -lpython3.4
 
     # ldconfig is being used to find libboost_python, but it's in a different
     # place in different distros (and is not in the default $PATH on Debian).
     # First, check to see if it's on the default $PATH.
-    system(which ldconfig > /dev/null) {
+    system(which ldconfig 2&> /dev/null) {
         LDCONFIG_BIN = "ldconfig"
     }
     # If that failed, then search for it in its usual places.
@@ -75,7 +75,7 @@ linux {
 
     # Check for different boost::python naming schemes
     LDCONFIG_OUT = $$system($$LDCONFIG_BIN -p|grep python)
-    for (b, $$list(boost_python-py34 boost_python3)) {
+    for (b, $$list(boost_python-py34 boost_python3 boost_python-3.4)) {
         contains(LDCONFIG_OUT, "lib$${b}.so") {
             LIBS += "-l$$b"
             GOT_BOOST_PYTHON = True
```

My changes aren't very big and it would be awesome if workarounds for different distributions could be integrated, so users of the biggest distros can build antimony out the box.

astronouth7303, 2015-06-06:
There has got to be a better way to deal with this...

jayay, 2015-06-07:
I've been thinking of something like

```bash
ls -1 /usr/bin/python* | grep -xP "^[/a-zA-Z]+(-config-3(\.4)?$|3(\.4)?-config$)?" | tail -n +2
```