mkeeter/antimony #79: Convex hull

angus-g, 2015-06-07:
I'm a fan of being able to generate a convex hull of multiple shapes. From what I can tell, this can't quite be replicated by CSG nodes at the moment.

astronouth7303, 2015-06-07:
Discussed on [the list](https://groups.google.com/d/topic/antimony-dev/ch8edLGItgQ/discussion). From @mkeeter:

> Interesting question.  There isn’t anything equivalent at the moment, because 
convex hulls fall into the category of “easy to do with b-reps, hard to do with 
implicit surfaces”. 
>
> If anyone has thoughts (or relevant research literature), I’d be interested in 
thinking more about this kind of operator.

Pupo1, 2015-11-12:
Perhaps the developers of Openscad might give a hint on how to do it?

astronouth7303, 2015-11-12:
The internal representation used by Antimony (functional representation) makes this non-trivial, as discussed in #31.

tl;dr: Antimony doesn't actually know the shape of a model, it knows a function that happens to be at 0 in all the places the model is. This makes certain operations easier (CSG) and certain operations harder (offsets, hulls).

aurelw, 2016-10-01:
Is it completely out of the question to introduce non-implicit surfaces for these kind of problems? Transform the functional representation into an SDF. SDF to convex hull is fairly straight forward.