mkeeter/antimony #80: Crashes when I try to change variable type in script

Highstaker, 2015-06-07:
Here's what we do to reproduce it.

* Create a node.
* Go to its script.
* Create an input, for example ```input('a',int)```. It works fine so far.
* Change the type to, say, float

We notice that the input/output handles vanish. And when I click where these handle squares are supposed to be - it crashes.

astronouth7303, 2015-06-07:
Hm. I got a crash testing this while in the middle of typing. More information coming.

As a note: The appearance of drag boxes on `int` lines is inconsistent. Dropping in my Parameters node shows them.

astronouth7303, 2015-06-07:
Hm. I can't reproduce this under a debug build....

@Highstaker what Git revision did you build? Should say in Help > About.

astronouth7303, 2015-06-07:
I'm unable to reproduce now.

Highstaker, 2015-06-07:
Git revision e20af58

https://github.com/mkeeter/antimony/commit/e20af58a8efa5a0e5c965dab4fb9f2b20ff81434

My system is Ubuntu 14.04

Tell me if you need more info.

I tried it on older revisions initially, retried now on the newest one, the problem persists.

astronouth7303, 2015-06-08:
Ok, got a backtrace.

```
#0  0x00007ffff52d7538 in QObjectPrivate::addConnection(int, QObjectPrivate::Connection*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#1  0x00007ffff52d7800 in QObjectPrivate::connectImpl(QObject const*, int, QObject const*, void**, QtPrivate::QSlotObjectBase*, Qt::ConnectionType, int const*, QMetaObject const*) ()
   from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#2  0x00007ffff52d7c78 in QObject::connectImpl(QObject const*, void**, QObject const*, void**, QtPrivate::QSlotObjectBase*, Qt::ConnectionType, int const*, QMetaObject const*) ()
   from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#3  0x0000000000727224 in QObject::connect<void (QObject::*)(QObject*), void (Datum::*)()> (sender=0x18bdde0, signal=
    (void (QObject::*)(QObject * const, QObject *)) 0x7ffff52d3d30 <QObject::destroyed(QObject*)>, receiver=0x0, 
    slot=(void (Datum::*)(Datum * const)) 0x7b7ec8 <Datum::connectionChanged()>, type=Qt::AutoConnection) at ../../../.local/Qt5.4.1/5.4/gcc_64/include/QtCore/qobject.h:239
        types = 0x0
#4  0x0000000000725520 in Datum::linkFrom (this=0x0) at ../src/graph/datum/datum.cpp:86
        link = 0x18bdde0
#5  0x00000000007878ab in OutputPort::mousePressEvent (this=0x1957eb0, event=0x7fffffffc910) at ../src/ui/canvas/port.cpp:73
        c = 0x1957ec0
#6  0x00007ffff668c817 in QGraphicsItem::sceneEvent(QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#7  0x00007ffff66bb0ad in ?? () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#8  0x00007ffff66c1faf in ?? () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#9  0x00007ffff66c252f in QGraphicsScene::mousePressEvent(QGraphicsSceneMouseEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#10 0x00007ffff66ca056 in QGraphicsScene::event(QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#11 0x00007ffff63852b4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#12 0x00007ffff6388c86 in QApplication::notify(QObject*, QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#13 0x00007ffff52a0e24 in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#14 0x00007ffff66debc0 in QGraphicsView::mousePressEvent(QMouseEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#15 0x0000000000774caa in Canvas::mousePressEvent (this=0x13ef2b0, event=0x7fffffffd330) at ../src/ui/canvas/canvas.cpp:66
No locals.
#16 0x00007ffff63c0b82 in QWidget::event(QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#17 0x00007ffff64c7cf6 in QFrame::event(QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#18 0x00007ffff66e817b in QGraphicsView::viewportEvent(QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#19 0x00007ffff52a108d in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#20 0x00007ffff638527f in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#21 0x00007ffff6388a4c in QApplication::notify(QObject*, QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#22 0x00007ffff52a0e24 in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
#23 0x00007ffff6387978 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) ()
   from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#24 0x00007ffff63dbb57 in ?? () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#25 0x00007ffff63de648 in ?? () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#26 0x00007ffff63852b4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#27 0x00007ffff6388c86 in QApplication::notify(QObject*, QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Widgets.so.5
No symbol table info available.
#28 0x00007ffff52a0e24 in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /home/astronouth7303/.local/Qt5.4.1/5.4/gcc_64/lib/libQt5Core.so.5
No symbol table info available.
[... cut ...]
              stringdata = 0x7f1fc0 <qt_meta_stringdata_App>, data = 0x7f2340 <qt_meta_data_App>, 
              static_metacall = 0x7ba28a <App::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, extradata = 0x0}}, graph_scene = 0xbb8990, 
          view_scene = 0xbb9ea0, filename = {static null = {<No data fields>}, d = 0x7ffff56e9880 <QArrayData::shared_null>}, root = 0xbb9ed0, stack = 0xbb9f70, network = 0xbba030}
        d = {static null = {<No data fields>}, d = 0xc2d220}
        fab = 0x7fffeeb80098
```

mkeeter, 2015-06-08:
I can get the boxes to disappear if I start with a `float`, change to an `int`, then change it back to an `int`.  Clicking on the (now-disappeared) box causes a crash.

Places to start looking are either in [`ScriptDatum::makeInput`](https://github.com/mkeeter/antimony/blob/develop/src/graph/datum/datums/script_datum.cpp#L49-L50)  or [`NodeInspector::populateLists`](https://github.com/mkeeter/antimony/blob/develop/src/ui/canvas/inspector/inspector.cpp#L138), but this is firmly in "deep magic" territory; I'll definitely be looking into it myself.

Highstaker, 2015-06-08:
Yeah, I tried with ```str``` and ```int``` - works fine. Same with ```str``` and ```float```. But not with ```int``` and ```float```

mkeeter, 2015-08-01:
I'm going to assume that this is fixed with the rewritten graph engine.  Let me know if it shows up again.

closed 2015-08-01 by mkeeter