mkeeter/antimony #87: Open GL problem can only see wireframe shapes

estocas, 2015-06-18:
I have install antimony in a Sony Vaio VPCF12Z1E running ubuntu 14.04 LTS and it's working. My Open GL drivers are working too.

When I open Antimony, I only can see wireframe shapes instead of white shapes. This is the message error that I can see in terminal:

Thanks in advance!

adolfo@adolfo-vaio:~/antimony/build$ ./antimony
QOpenGLShader::compile(Fragment): 0(14) : error C7537: OpenGL does not allow 'flat' after a type specifier
0(14) : error C7538: OpenGL does not allow 'flat' after 'uniform'
0(26) : error C0000: syntax error, unexpected "==", expecting "::" at token "=="
0(31) : error C0000: syntax error, unexpected reserved word "else" at token "else"
0(54) : error C0000: syntax error, unexpected '=', expecting "::" at token "="

*** Problematic Fragment shader source code ***
#version 120
#define lowp
#define mediump
#define highp

varying vec2 texture_coord;
uniform sampler2D depth_tex;
uniform sampler2D shaded_tex;

uniform float zmin_local;
uniform float dz_local;
uniform float zmin_global;
uniform float dz_global;
uniform int flat;

uniform vec3 color;

vec4 shade(vec4 norm)
{
    vec3 light = vec3(0.99 * color.r, 0.96 * color.g, 0.89 * color.b);
    vec3 dark = vec3(0.20 * color.r, 0.25 * color.g, 0.3 * color.b);

    float a = dot(2 * (norm.xyz - vec3(0.5)), vec3(0.0, 0.0, 1.0));
    float b = dot(2 * (norm.xyz - vec3(0.5)), vec3(0.57, 0.57, 0.57));

    if (flat == 1)
    {
        a = a*0.5 + 0.5;
        return vec4((a*light + (1-a)*dark), 1);
    }
    else
    {
        return vec4((a*light + (1-a)*dark)*0.35 +
                    (b*light + (1-b)*dark)*0.65, 1);
    }
}

void main() {
    vec4 depth = texture2D(depth_tex, texture_coord);

    if (depth.r == 0.0f)
    {
        discard;
    }

    gl_FragColor = shade(texture2D(shaded_tex, texture_coord));

    // Due to the way that Qt renders a GraphicsScene
    // (with OpenGL acceleration), we're limited to the
    // depth range of (0.5, 1].  In practice, we'll upse
    // [0.6, 0.9]
    float fd_local = depth.r * dz_local + zmin_local;
    float fd_global = (fd_local - zmin_global) / dz_global;
    gl_FragDepth = 0.9 - fd_global*0.3;
}

***

mkeeter, 2015-06-19:
Hah, it turns out `flat` is an OpenGL keyword (I was using it as a variable name).
This should be fixed in 026c50186e700936bccb39d10d65fe28f4477202 -- can you pull and confirm?

toomuchcookies, 2015-06-19:
For me the shading view is not working either. 
Last commit it worked was b56ad9dae4645a5f57fd0c0a80172696cea1793a .
In commit 63ece77 only the height map was working.

Tibzor, 2015-06-20:
Still no shading view when rotating with commit 1d06401, but it seems to works when using preset 2D views (front, rear, top...). Switches to wireframe when trying to rotate. No feedback on terminal though...

toomuchcookies, 2015-06-20:
I agree with Tibzor. Also, i would like to add, that even changing some values makes the shading disappear. Testing with a sphere (0,0,0,1). If i decrease the radius below 1.0 the shading disappears. I was able to get it back with a radius of 0.9999999. And as mentioned, it only is drawn for specific perspectives. Please advise.

mkeeter, 2015-06-21:
@toomuchcookies , which commit are you testing?  Can you upload or send me a few screenshots?

toomuchcookies, 2015-06-21:
@mkeeter : I'll do you one better. Here is a link to a screencast i did:
https://www.youtube.com/watch?v=oDUTpt2HeCw
As you can see in the video, i'm using the latest git revision 1d06401 . There are no errors in the command line (from where i've started Antimony). I think there is a problem with float as an input to the shading routine?

mkeeter, 2015-06-21:
That's very bizarre.  The only thing that I'm suspicious of is [sneaky struct packing tricks](https://github.com/mkeeter/antimony/blob/452037112c833c6f522ce47389c990302b5fcd9b/lib/fab/src/tree/math/math_g.c#L6-L7) in gradient calculations; that's removed in the latest commit (4526e839dd27e65d71547a799525c83e253b2e41).

If that doesn't fix it for you, the next step would be making a testing branch that detects if `NaN`s are being introduced into gradient calculations.

toomuchcookies, 2015-06-21:
Up until 4520371 it is the same. But the commit 4526e83 actually made the shading go away. Now, i can't see any shading or height map anymore.

mkeeter, 2015-06-21:
For some reason, I'm now able to reproduce this.  Bisecting now...

toomuchcookies, 2015-06-21:
You'll tell me, if i need to do anything.

mkeeter, 2015-06-21:
Now that I can reproduce locally I should have a much easier time tracking this down.
Thanks for the help thus far.

mkeeter, 2015-06-21:
07f0c6dd25d71aa9e372b442ee655542b955bf0a fixes an obvious bug that was recently introduced.  Can you see if that solves your issue?

I've found that Qt isn't great at doing rebuilds correctly; please rebuild by deleting the entire build directory and starting over (from `qmake ../sb.pro`).

toomuchcookies, 2015-06-21:
Now it's back to how it was before, (like in the screencast).

mkeeter, 2015-06-22:
Okay, I'm back to being mystified.

I've made a branch named [`debug/missing-shapes`](https://github.com/mkeeter/antimony/tree/debug/missing-shapes) that will verbosely print details on the render process; can you switch branches and see what's printed to the console?

toomuchcookies, 2015-06-22:
Ok, so i first started antimony, inserted a sphere and just changed the perspective. As always, when i changed the perspective, the rendering went away.
https://gist.github.com/toomuchcookies/25954f02572573bd8de6

Then i started antimony and with the default perspective i inserted a sphere again and changed the radius of the sphere:
https://gist.github.com/toomuchcookies/65d5b0d0ff12fe2a5bf5

Then i created two spheres, one with radius 1 and the other with a radius of 0.5, built a union and then exported them to stl. I then loaded the file into blender and could only see one sphere. I am guessing the problem is not with the render, it is with the fab engine itself? 

Next test: I created only one sphere radius 1, then i put it's center coordinate to some fraction (y=.3) then exported it to stl and loaded it into blender and as i was expecting, the bounding box of the sphere was moved, but not the sphere itself.

Hope you can do something with this info.

Tibzor, 2015-06-22:
Don't know if you can use another example for this problem...
But I've tried commit cec2d6c and here is a screencast I made : https://youtu.be/uAQhd7o7C7A
And the log file (sorry for the 18K lines...) : https://gist.github.com/Tibzor/d548a32e8a6c891c7d61
As you can see, there's some pretty weird behavior from the second disc !

Hope it helps

toomuchcookies, 2015-06-26:
Any news about this issue?? I would like to help, but until now i didn't find the function that transforms a functional Shape to polygons.

mkeeter, 2015-06-27:
No luck on this, because I can't reproduce it at all on my system.

Does your system have the option to install binary drivers?

toomuchcookies, 2015-06-29:
@mkeeter : As far as i know (not home right now), i have the default (linux mint) nvidia binary driver installed. I guess i can install another one, if you know which one to try. 
Other thought: I thought you were already able to reproduce the problem?

mkeeter, 2015-06-29:
I thought I was reproducing the problem in 4526e839dd27e65d71547a799525c83e253b2e41, but I had actually introduced an unrelated bug (fixed in 07f0c6dd25d71aa9e372b442ee655542b955bf0a).

toomuchcookies, 2015-06-29:
Ok, so what driver should i be testing?

mkeeter, 2015-06-29:
No idea; a bit of searching found [this article](http://www.binarytides.com/install-nvidia-drivers-linux-mint-16/) which walks through the installation process for binary drivers.

toomuchcookies, 2015-06-29:
Ok, now i understand. I thought you had a specific driver in mind. I just tried all available drivers for my system (nouveau, nvidia-331 and nvidia-304) but there is no difference there. 

Since i have the suspicion that the bug has to do with either the parser or the triangulation step (remember, the same bug can be seen, when outputting stl) i was poking around and trying to get some debugging output, but i have no clue where to find what (A nice overview over what bit of code goes where would be appreciated). Specifically, i tried this: d2e94f7 . You can see my annotations, specially the one in get_float in the parser code, which i was expecting to output something somewhere but it doesn't..

mkeeter, 2015-06-29:
Something looks very strange with the [git history](https://github.com/mkeeter/antimony/network) -- it looks like you branched off two weeks ago and haven't been keeping up with `develop`, so you've been missing a bunch of changes (including 4526e839dd27e65d71547a799525c83e253b2e41, which was a fix specifically targetted at this issue).  Can you `reset --hard` to the current state of `develop` (or use some other strategy to get up to date)?

toomuchcookies, 2015-06-29:
@mkeeter : I run two git repos locally, one is a clone of your repo, where i test the commits and build the regular antimony, the other is the one i use to try to debug and it's a clone of my fork of your repo. So, the tests i did before are still valid.

toomuchcookies, 2015-07-05:
@mkeeter : sorry to bug you, but could you please address the issue i'm having with the output code. I'd really like to use antimony, but this issue is making it unusable for me. Based on @Tibzor 's screencast, it is not only happening to me. It would help alot, if you could point me to the functions where the parsing and/or triangulation is done. Thanks.

mkeeter, 2015-07-05:
Like I said, I can't reproduce this on any of the machines that I have available to me (Mac OS laptop, Linux VM, Chromebook running Linux using Crouton).

Parsing is here:
https://github.com/mkeeter/antimony/blob/develop/lib/fab/src/tree/parser.c#L92-L119

Mesh export is here:
https://github.com/mkeeter/antimony/blob/develop/app/src/export/export_mesh.cpp

toomuchcookies, 2015-07-05:
@fros1y : I was able to narrow down the problem to this commit https://github.com/mkeeter/antimony/commit/54e72de55e46669797d7fdc8641f8956d9d5b1a5 where you introduce the v2 parser. Can you give me any guidance as to how to debug this? Right now i am unsure, how the new parser distinguishes between different syntaxes, i am trying to get into it, but any hints would be appreciated. 

If unclear, please have a look at the above two screen casts.

mkeeter, 2015-07-05:
@fros1y, any idea why the new parser would cause bizarre behavior (shapes not appearing, etc) on certain systems?  I haven't been able to reproduce it myself, but there are at least two cases in the wild.

toomuchcookies, 2015-07-05:
Ok, so i think i know what the problem is and why it only happens to some of us. It seems it has to do with locale and regionality. I am (was) running linux with German locale which makes the decimal seperator a comma instead of a dot and then atof, which is used in v2syntax.y misbehaves. I was able to confirm that just by changing my locale (to GB) i was able to run antimony very normally..

fros1y, 2015-07-05:
Thanks, that's a great catch!  One of the parser changes was to move to using atof() instead of a custom function Matt was maintaining -- which obviously did not honor locale and actually worked correctly.

toomuchcookies, 2015-07-06:
Does anyone know of a better way than changing the locale? I am not really sure, if what i've done will break something else along the way. Or maybe we can switch back to the function that @mkeeter was using before? Any opinions?

mkeeter, 2015-07-06:
I think that changing the locale is a fine solution with a bit of checking to make sure that nothing else breaks.  I'm already changing it in C++ [here](https://github.com/mkeeter/antimony/blob/develop/app/src/app/main.cpp#L18-L19) to enable Unicode support.

mkeeter, 2015-07-08:
Okay, I've tested this out and it appears to work fine.

Let me know if you have any further issues (and thanks again for your detective work).

closed 2015-07-08 by mkeeter