mkeeter/antimony #91: Compile error: recipe for target 'sub-app-make_first-ordered' failed

FherraZ, 2015-06-21:
Hi guys, while compiling on Ubuntu Mate 15.04 I got the following error:

mkdir -p /home/fherraz/antimony/build/app/sb
cp -f -R /home/fherraz/antimony/app/../py/nodes /home/fherraz/antimony/build/app/sb
cp -f -R /home/fherraz/antimony/app/../py/fab /home/fherraz/antimony/build/app/sb
src/graph/datum/types/eval_datum.o: file not recognized: File truncated
collect2: error: ld returned 1 exit status
Makefile:550: recipe for target 'Antimony' failed
make[1]: *** [Antimony] Error 1
make[1]: Leaving directory '/home/fherraz/antimony/build/app'
Makefile:86: recipe for target 'sub-app-make_first-ordered' failed
make: *** [sub-app-make_first-ordered] Error 2

I am assuming I was hitting the end of the compile because it was working well and it took its time, now every time I use make -j8 command it gives me the above error.

Thank you.

mkeeter, 2015-06-21:
As a first step, I'd try cleaning up the build directory and re-building.

FherraZ, 2015-06-22:
Ok thank you, I will do tonight and post any changes

FherraZ, 2015-06-22:
You where right, compiled without error, thank you.

closed 2015-06-22 by FherraZ