mkeeter/antimony #100: ImportError: No module named 'fab'

Sass-Meister, 2015-07-06:
After installing antimony on Ubuntu in a vm on Windows 7. When I try and run it a I get a pop up with this message:

Import Error:

Could not find fab Python module.
Antimony will now exit

And in the shell I have this text:
ImportError: No module named 'fab'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 63, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 12, in <module>
    import subprocess, tempfile, os.path, re, pwd, grp, os, time
  File "/usr/lib/python3.4/tempfile.py", line 30, in <module>
    import functools as _functools
  File "/usr/lib/python3.4/functools.py", line 22, in <module>
    from types import MappingProxyType
ImportError: cannot import name 'MappingProxyType'

I already tried 'pip install fab' and 'pip install fabric' but neither of those worked. How can I fix this?

mkeeter, 2015-07-06:
[Here's](https://github.com/mkeeter/antimony/issues/21) a similar issue.

Following the same debugging procedure:
Are you running it from the build directory, or did you run `make install`?

Normally, Antimony will produce a folder structure `sb/fab/...` that contains the standard shapes library and a few magic hooks. Do you see that folder?

Sass-Meister, 2015-07-06:
I am running it in the build folder, and I see '''~/Desktop/antimony/build/app/sb/fav''' and it has '''__init___.py shapes.py types.py'''

mkeeter, 2015-07-07:
Continuing to copy-paste from the other thread:

It looks like there's a conflict between `fab.types` and the system's `types` module.  I'm able to reproduce this by running `python3` from the `app/sb/fab` folder then trying `import functools`, but not sure why it's happening when you run Antimony from the build directory.

Silly question -- do you see `types.py` in the build directory (from where you're running Antimony) or the `app` folder?  If you just run `python3 -c "import functools"` from the folder from where you're running Antimony, does it fail in the same way?

dylanmc, 2015-08-10:
I'm having this problem with the pre-built MacOS package...Hmmm.

mkeeter, 2015-08-11:
@dylanmc, the `fab` folder should be inside the Antimony bundle at `Antimony.app/Contents/Frameworks/Python.framework/Versions/3.4/lib/python3.4/fab`.  Can you see if that folder exists and what it contains?

dylanmc, 2015-08-11:
@mkeeter - whoops - I goofed, I must have moved my mis-built app into /Applications. Fresh download worked fine.

mkeeter, 2015-08-11:
Cool, glad everything is working now.

closed 2015-08-11 by mkeeter