mkeeter/antimony #106: RuntimeError: Cannot declare multiple UI elements on same line.

joshuashort, 2015-07-21:
This appeared while trying to make a polygon node that provides a handle on each of its vertices..

```
import fab

title('Polygon with Anchors')

input('x', float)
input('y', float)
input('N', int)
input('r', float)

output('shape', fab.shapes.move(fab.shapes.polygon_radius(x, y, r, N), x, y))

a_ = 0 if (N % 2) else (math.pi / N)

vertices = [
	(x + r * math.sin(a_ + a * math.pi * 2/N),
	 y + r * math.cos(a_ + a * math.pi * 2/N), 0)
	for a in range(N)]

fab.ui.wireframe(vertices, close=True)

fab.ui.point(x, y)
	
for vertex_n, vertex in enumerate(vertices):
	output("vertex_{0}_x".format(vertex_n), vertex[0])
	output("vertex_{0}_y".format(vertex_n), vertex[1])

	#fab.ui.point(vertex[0], vertex[1], drag=(None, r, None))
```

Uncommenting the last line produces the error. The message is generated by https://github.com/mkeeter/antimony/blob/203611d81447d3e8b0ae9bf92eb8e9df52f1fc35/app/src/graph/hooks/ui.cpp#L92-L112.

This also doesn't work:
```
def trying_this(vertex):
	fab.ui.point(vertex[0], vertex[1], drag=(None, r, None))

for vertex_n, vertex in enumerate(vertices):
	output("vertex_{0}_x".format(vertex_n), vertex[0])
	output("vertex_{0}_y".format(vertex_n), vertex[1])
	trying_this(vertex)
```

Although that examples fails with ```SyntaxError: unexpected EOF while parsing```

It works when the for-loop is only dealing with a single-item list (e.g. ```for vertex_n, vertex in enumerate(vertices[:1]):```)

mkeeter, 2015-07-21:
UI elements are ID'd by their line number as a way of recognizing them from one script evaluation to the next, so this is an intentional limitation -- otherwise, the system that caches UI handles wouldn't be able to distinguish between these points.

I'll look into removing this limitation once the new graph engine is merged into `develop`.

mkeeter, 2016-06-26:
In the `develop` branch, you can now pass in `key=?` keyword argument to both `point` and `wireframe` to disambiguate between multiple calls on the same line, e.g.

```
for i in range(10):
    sb.ui.point(i, 0, key=i)
```

(props to [dear imgui](https://github.com/ocornut/imgui) for this technique)

closed 2016-06-26 by mkeeter