mkeeter/antimony #113: Transparent background on View window.

ApostolosB, 2015-08-06:
This happens on Linux with wayland and an intel i915 card. Tried on 2 compositors. Weston and on enlightenment from git.

wayland 1.8.1
qt 5.5

I set QT_QPA_PLATFORM=wayland-egl

On Xwayland it works as it should.

mkeeter, 2015-08-07:
Interesting.  Are you running open-source or binary drivers?

ApostolosB, 2015-08-07:
Open. Intel doesn't have closed drivers for its cards on linux so i can't test with a different driver. 

One more thing i noticed -in case it can give you a clue- is that if i select View>Height map it creates a black background just around the object and leaves the rest of the window transparent.

mkeeter, 2015-08-11:
Okay, that's pretty funky.  Can you mess with the clear color [here](https://github.com/mkeeter/antimony/blob/develop/app/src/ui/viewport/viewport.cpp#L587-L588) and see if that makes a difference?

ApostolosB, 2015-08-12:
Ok. Changed  glClearColor(0.0, 0.0, 0.0, 0.0); to  glClearColor(0.0, 0.0, 0.0, 1.0); and now i get a black background. 

Didn't test if it breaks something else but seems to work on wayland (both compositors) and xwayland.

So yeah this fixes it.

mkeeter, 2015-08-13:
Aha!  For some reason, I was under the impression that the last value was a depth clearing value; it's actually alpha, so that makes perfect sense.  Thanks for doing that test -- would you mind opening up a pull request into `develop` that makes this change?

mkeeter, 2015-08-14:
Fixed by #115

closed 2015-08-14 by mkeeter