mkeeter/antimony #114: .node files installation

sagitter, 2015-08-07:
Hi all,

what does https://github.com/mkeeter/antimony/tree/develop/py/nodes directory contain?
And where all ```.node``` files should be installed in Linux system to be useful for ```Antimony```?

mkeeter, 2015-08-07:
The nested lists of `.node` files are populated into the Add menu; they're a bunch of Python scripts that define basic shapes and transforms.

You shouldn't need to do anything with these folders manually; Antimony knows where to find them (when running the executable `$BINDIR/antimony`, it will search `$BINDIR/sb/nodes`). 

They should be installed with (`sudo`) `make install` into `/usr/local/bin/sb/nodes` (with the executable ending up in `/usr/local/bin/Antimony`); if you're just running `Antimony` from the build directory, they should be in `$BUILDDIR/app/sb/nodes`.  Both of these copying operations happen automatically in the build scripts.

sagitter, 2015-08-07:
Hi,

for an official packaging it's not a good idea, i think.

Could they be stored under /usr/share and redirect antimony searching against new location?

Thanks.

mkeeter, 2015-08-07:
The relevant part of the install script is [here](https://github.com/mkeeter/antimony/blob/develop/app/app.pro#L42-L48).

We already search for nodes in two places: [`bundledNodePath`](https://github.com/mkeeter/antimony/blob/develop/app/src/app/app.cpp#L334-L361) and [`userNodePath`](https://github.com/mkeeter/antimony/blob/develop/app/src/app/app.cpp#L363-L369) (which uses the [QStandardPaths](http://doc.qt.io/qt-5/qstandardpaths.html) class, producing `~/.local/share/Antimony` on Linux).

These two locations are used [here](https://github.com/mkeeter/antimony/blob/develop/app/src/ui/main_window.cpp#L172-L19) in `populateNodeMenu`.

Want to propose a new organization structure and submit a PR?

sagitter, 2015-08-08:
Installing nodes paths under ```/usr/share``` could already be a good choice; however, i wouldn't be able to change ```Antimony``` Qt code.

Thanks.

sagitter, 2015-08-10:
>We already search for nodes in two places: bundledNodePath and userNodePath (which uses the QStandardPaths class, producing ~/.local/share/Antimony on Linux).

How do i set ```QStandardPaths::AppDataLocation``` to ```/usr/share/<appname>``` ?

astronouth7303, 2015-08-12:
@sagitter You can't modify them.

The correct thing to do is to use the full directory list for `QStandardPaths::AppDataLocation`. This is `~/.local/share/<APPNAME>`, `/usr/local/share/<APPNAME>`, and `/usr/share/<APPNAME>`.

mkeeter, 2015-08-12:
Interesting -- that would involve calling `QStandardPaths::standardLocations` instead of just `QStandardPaths::writableLocation`, but seems perfectly reasonable.

sagitter, 2015-08-12:
@astronouth7303,

do you mean

```nodes_folder.path = /usr/share/Antimony/nodes```
```nodes_folder.files = /usr/share/Antimony/nodes/*```

?

astronouth7303, 2015-08-19:
Not exactly. Refactor to loop through a `QStringList`, as returned from the previously-mentioned `QStandardPaths::standardLocations()`.

This will do the Right Thing for every platform, with all the possible locations to look in.

sagitter, 2015-09-27:
> @sagitter You can't modify them.
The correct thing to do is to use the full directory list for QStandardPaths::AppDataLocation. This is ~/.local/share/<APPNAME>, /usr/local/share/<APPNAME>, and /usr/share/<APPNAME>.

Please, can you help me to patch this variable?

mkeeter, 2016-07-06:
Okay, this should be fixed in the `release/0.9.2b` branch
(to be merged into `develop` after the release is complete).

Here are the paths that are searched by default:
[image]
(where `Filtered` paths are the ones that actually exist)

closed 2016-07-06 by mkeeter

ApostolosB, 2017-08-19:
I fresh installed latest git and the ./local/share/nodes dir wasn't created. Is it a known issue? (Linux)