mkeeter/antimony #119: Exporting a heightmap crashes the program

tenedor, 2015-09-14:
Steps to reproduce:
- create a `Heightmap (.png)` export node and connect an input shape
- click the `Export` button that appears
- select a resolution and file location and hit `Save`

A progress bar dialog pops up that says "Exporting..." but within a few seconds the program crashes. An empty .png file will be created in the intended file location.

I am running OS X Yosemite 10.10.2 using the Antimony 0.9.0 direct-download application.

Crash log:

Process:               antimony [34700]
Path:                  /Applications/Antimony.app/Contents/MacOS/antimony
Identifier:            com.impraxical.antimony
Version:               0.9.0c
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           antimony [34700]
User ID:               502

Date/Time:             2015-09-14 16:52:28.535 -0400
OS Version:            Mac OS X 10.10.2 (14C1510)
Report Version:        11
Anonymous UUID:        75191827-6486-EF63-DB8C-5B66390F3E04

Sleep/Wake UUID:       614D64F0-7E90-473F-858D-710E7108A8E2

Time Awake Since Boot: 140000 seconds
Time Since Wake:       14000 seconds

Crashed Thread:        11  QThread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000

Thread 0:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreFoundation      	0x00007fff93b8e4f1 -[__NSArrayM countByEnumeratingWithState:objects:count:] + 1
1   com.apple.AppKit              	0x00007fff8b4aea24 -[NSWindow _clearBackingStoreForBackdropViews] + 230
2   com.apple.AppKit              	0x00007fff8aed8a02 -[NSWindow flushWindow] + 120
3   com.apple.AppKit              	0x00007fff8aead9ef -[NSView _displayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:] + 4589
4   com.apple.AppKit              	0x00007fff8ae8be2d -[NSView displayIfNeeded] + 1876
5   com.apple.AppKit              	0x00007fff8aea91c5 -[NSWindow displayIfNeeded] + 232
6   com.apple.AppKit              	0x00007fff8aee6322 _handleWindowNeedsDisplayOrLayoutOrUpdateConstraints + 936
7   com.apple.AppKit              	0x00007fff8b4af0e1 __83-[NSWindow _postWindowNeedsDisplayOrLayoutOrUpdateConstraintsUnlessPostingDisabled]_block_invoke1531 + 46
8   com.apple.CoreFoundation      	0x00007fff93ba1da7 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
9   com.apple.CoreFoundation      	0x00007fff93ba1d00 __CFRunLoopDoObservers + 368
10  com.apple.CoreFoundation      	0x00007fff93b93e08 __CFRunLoopRun + 872
11  com.apple.CoreFoundation      	0x00007fff93b93858 CFRunLoopRunSpecific + 296
12  com.apple.HIToolbox           	0x00007fff90dafaef RunCurrentEventLoopInMode + 235
13  com.apple.HIToolbox           	0x00007fff90daf76e ReceiveNextEventCommon + 179
14  com.apple.HIToolbox           	0x00007fff90daf6ab _BlockUntilNextEventMatchingListInModeWithFilter + 71
15  com.apple.AppKit              	0x00007fff8ad86f81 _DPSNextEvent + 964
16  com.apple.AppKit              	0x00007fff8ad86730 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 194
17  com.apple.AppKit              	0x00007fff8b08bf67 -[NSApplication _realDoModalLoop:peek:] + 666
18  com.apple.AppKit              	0x00007fff8b0c51f4 -[NSApplication runModalSession:] + 71
19  libqcocoa.dylib               	0x000000010818123a 0x108161000 + 131642
20  org.qt-project.QtCore         	0x000000010617237d QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
21  org.qt-project.QtWidgets      	0x00000001055c9192 QDialog::exec() + 514
22  com.impraxical.antimony       	0x0000000104f8b11d ExportHeightmapWorker::run() + 2221
23  org.qt-project.QtWidgets      	0x00000001056d2876 QGraphicsItem::sceneEvent(QEvent*) + 534
24  org.qt-project.QtWidgets      	0x00000001056dbab4 QGraphicsTextItem::sceneEvent(QEvent*) + 148
25  org.qt-project.QtWidgets      	0x00000001056f9882 0x1053b3000 + 3434626
26  org.qt-project.QtWidgets      	0x00000001056fb25a 0x1053b3000 + 3441242
27  org.qt-project.QtWidgets      	0x00000001056fbab1 0x1053b3000 + 3443377
28  org.qt-project.QtWidgets      	0x0000000105705d9a QGraphicsScene::mousePressEvent(QGraphicsSceneMouseEvent*) + 90
29  org.qt-project.QtWidgets      	0x0000000105700c87 QGraphicsScene::event(QEvent*) + 1927
30  org.qt-project.QtWidgets      	0x00000001053e432b QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
31  org.qt-project.QtWidgets      	0x00000001053e7648 QApplication::notify(QObject*, QEvent*) + 8136
32  org.qt-project.QtCore         	0x0000000106174d83 QCoreApplication::notifyInternal(QObject*, QEvent*) + 115
33  org.qt-project.QtWidgets      	0x00000001053e8c3d 0x1053b3000 + 220221
[... 475 lines cut ...]
__UNICODE                           544K
mapped file                        71.6M
shared memory                        68K
===========                      =======
TOTAL                             648.9M
TOTAL, minus reserved VM space    648.4M
 

Model: MacBookPro8,1, BootROM MBP81.0047.B27, 2 processors, Intel Core i5, 2.4 GHz, 8 GB, SMC 1.68f99
Graphics: Intel HD Graphics 3000, Intel HD Graphics 3000, Built-In, 512 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1333 MHz, 0x80AD, 0x484D54333531533642465238432D48392020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1333 MHz, 0x80AD, 0x484D54333531533642465238432D48392020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xD6), Broadcom BCM43xx 1.0 (5.106.98.100.24)
Bluetooth: Version 4.3.2f6 15235, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: TOSHIBA MK5065GSXF, 500.11 GB
Serial ATA Device: MATSHITADVD-R   UJ-8A8
USB Device: Hub
USB Device: IR Receiver
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 25.1

mkeeter, 2015-09-14:
I've added debug print statements to the png export and put a debug build online [here](http://www.mattkeeter.com/misc/Antimony.dmg).

Can you try to reproduce the issue with this build?

Running it from the Terminal (i.e. by calling `./Antimony.app/Contents/MacOS/Antimony`) will print out `libpng`'s error messages.

tenedor, 2015-09-14:
Here's the [https://gist.github.com/tenedor/67cc7f41539687e7f829](crash log) for running it with the debug build. Was it supposed to print libpng errors to terminal though? There was only a single line of output in my terminal:
```
$ ~/Documents/Antimony.app/Contents/MacOS/Antimony 
zsh: illegal hardware instruction  ~/Documents/Antimony.app/Contents/MacOS/Antimony
```

mkeeter, 2015-09-15:
Interesting.  Can you open up Activity Monitor while Antimony is running and paste in the list of open files and ports?

I'm suspicious that it's getting confused between the system's `libpng` (in `/System/Library/Frameworks/ImageIO.framework/Versions/A/Resources/libPng.dylib`) and the bundled `libpng`, but not sure why this doesn't happen on my machine.

mkeeter, 2015-09-15:
Also, I've posted another debug build at the [same URL](http://mattkeeter.com/misc/Antimony.dmg).

It should print out a debug statement with information about the png export, e.g.
```
~/code/antimony/build-release $ ./app/Antimony.app/Contents/MacOS/Antimony 
Exporting a 510 x 510 image to '/Users/mkeeter/Desktop/out.png' with pixel pointer 0x7ffb059a4200
```

tenedor, 2015-09-15:
Here's the [https://gist.github.com/tenedor/8f58ee3dfa3e14feda72](list of open files and the new crash log). Terminal output was

```
$ ~/Documents/Antimony.app/Contents/MacOS/Antimony
Exporting a 510 x 510 image to '/Users/daniel/Documents/Harvard/Semester 7/MAS 863/keyset.png' with pixel pointer 0x7ff47a374000
zsh: illegal hardware instruction  ~/Documents/Antimony.app/Contents/MacOS/Antimony
```

mkeeter, 2015-09-15:
This is getting very mysterious.

I was hopeful that the spaces in the file path were causing the crash, but tried it locally and nothing went wrong.

Here's the next step I'd like you to try:
Install `libpng` with [Homebrew](http://brew.sh/) (i.e. `brew install libpng`), then copy the newly-installed `/usr/local/lib/libpng16.16.dylib` into the Antimony app bundle (`Antimony.app/Contents/Frameworks/libpng16.16.dylib`), replacing the existing library in that location.

This should tell us if there's something wrong with the version of libpng that I'm distributing.

Thanks for your patience in debugging this issue.  It's hard when I can't reproduce something locally, so I appreciate your remote-debugging help.

mkeeter, 2015-09-15:
The relevant line when you disassemble `libpng` is

```
000000000001df19        bextrl  %eax, %r12d, %eax
```

A bit of searching finds `bextrl` is probably part of the [Bit Manipulation Instruction Set 1](https://en.wikipedia.org/wiki/Bit_Manipulation_Instruction_Sets#BMI1_.28Bit_Manipulation_Instruction_Set_1.29), though I can't find it anywhere other than compiler test suites...

Could it be that compiling `libpng` on my machine has chosen instructions that aren't available on your CPU?

mkeeter, 2015-09-15:
I've put up a new testing build at the [same URL](http://mattkeeter.com/misc/Antimony.dmg) that doesn't have the `bextrl` calls.  Maybe this will resolve your issue?

tenedor, 2015-09-16:
Thanks for all your help @mkeeter! It looks like it'll be a couple days before I can get back to you with more info - as of two hours ago my computer's hard disk appears permanently broken (the computer's four years old). I'll get back to this once I fix that.

tenedor, 2015-09-19:
Fantastic, the new code succeeds - I can now export height maps. (It worked without any updates to libpng.) Thanks!

mkeeter, 2015-09-19:
Cool, I'll switch to the friendlier libpng in the released disk image.  Thanks for you help in debugging this!

closed 2015-09-19 by mkeeter