mkeeter/antimony #121: Mac 10.6.8 build

OleksiyPuzikov, 2015-09-27:
Can subj be produced along with the version for newer MacOS'es?

Thanks in advance.

mkeeter, 2015-09-27:
Does the current release not work on Mac OS 10.6.8?
If so, in what way does it fail?

OleksiyPuzikov, 2015-09-27:
Unfortunately it doesn't. The error message is typical for something that's compiled for 10.7 only. Or is it?

8<---

Check with the developer to make sure Antimony works with this version of Mac OS X. 

Process:         antimony [25680]
Path:            /Volumes/Antimony 0.9.0c/Antimony.app/Contents/MacOS/antimony
Identifier:      com.impraxical.antimony
Version:         ??? (???)
Code Type:       X86-64 (Native)
Parent Process:  launchd [95]

Date/Time:       2015-09-27 21:48:35.615 +0300
OS Version:      Mac OS X 10.6.8 (10K549)
Report Version:  6

Interval Since Last Report:          118884 sec
Crashes Since Last Report:           1
Per-App Crashes Since Last Report:   1
Anonymous UUID:                      20C54114-4724-4ADE-91C4-AA306C176EE4

Exception Type:  EXC_BREAKPOINT (SIGTRAP)
Exception Codes: 0x0000000000000002, 0x0000000000000000
Crashed Thread:  0

Dyld Error Message:
  Library not loaded: /usr/lib/libc++.1.dylib
  Referenced from: /Volumes/Antimony 0.9.0c/Antimony.app/Contents/MacOS/antimony
  Reason: image not found

Binary Images:
    0x7fff5fc00000 -     0x7fff5fc3be0f  dyld 132.1 (???) <29DECB19-0193-2575-D838-CF743F0400B2> /usr/lib/dyld

Model: MacBook4,1, BootROM MB41.00C1.B00, 2 processors, Intel Core 2 Duo, 2.4 GHz, 4 GB, SMC 1.31f1
Graphics: Intel GMA X3100, GMA X3100, Built-In, 144 MB
Memory Module: global_name
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x88), Broadcom BCM43xx 1.0 (5.10.131.42.4)
Bluetooth: Version 2.4.5f3, 2 service, 12 devices, 1 incoming serial ports
Network Service: AirPort, AirPort, en1
Serial ATA Device: Samsung SSD 850 EVO 500GB, 465.76 GB
Parallel ATA Device: MATSHITADVD-R   UJ-857E
USB Device: Built-in iSight, 0x05ac  (Apple Inc.), 0x8501, 0xfd400000 / 2
USB Device: Bluetooth USB Host Controller, 0x05ac  (Apple Inc.), 0x8205, 0x1a100000 / 2
USB Device: Apple Internal Keyboard / Trackpad, 0x05ac  (Apple Inc.), 0x0229, 0x5d200000 / 3
USB Device: IR Receiver, 0x05ac  (Apple Inc.), 0x8242, 0x5d100000 / 2

mkeeter, 2015-09-27:
Unfortunately, 10.6 doesn't ship with libstdc++, which is needed because Antimony uses C++11.

You're welcome to try building it yourself -- [this StackOverflow discussion](http://stackoverflow.com/questions/14494513/how-do-i-get-back-c0x-c11-support-for-mac-os-x-10-6-deployment-using-xcode-4) maybe helpful -- but I'm not going to be officially supporting 10.6.

closed 2015-09-27 by mkeeter

OleksiyPuzikov, 2015-09-27:
Thanks for investigating this for me.